SpikeInterface/spikeinterface #2266: tridesclous2 crashes if svd/n_components != 6

rat-h, 2023-11-27:
If the parameter `svg["n_components"] != 6` sorting returns an error:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 246, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 205, in _process_chunk
    return [fn(*args) for args in chunk]
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 205, in <listcomp>
    return [fn(*args) for args in chunk]
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 437, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/node_pipeline.py", line 520, in _compute_peak_pipeline_chunk
    node_output = node.compute(traces_chunk, *node_input_args)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sortingcomponents/waveforms/temporal_pca.py", line 230, in compute
    projected_temporal_waveforms = self.pca_model.transform(temporal_waveforms)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/utils/_set_output.py", line 142, in wrapped
    data_to_wrap = f(self, X, *args, **kwargs)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/decomposition/_truncated_svd.py", line 287, in transform
    X = self._validate_data(X, accept_sparse=["csr", "csc"], reset=False)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/base.py", line 558, in _validate_data
    self._check_n_features(X, reset=reset)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/sklearn/base.py", line 359, in _check_n_features
    raise ValueError(
ValueError: X has 258 features, but TruncatedSVD is expecting 60 features as input.
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 254, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/internal/tridesclous2.py", line 178, in _run_from_folder
    output = run_node_pipeline(
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/node_pipeline.py", line 469, in run_node_pipeline
    processor.run()
  File "/home/rth/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/job_tools.py", line 399, in run
    for res in results:
  File "/usr/lib/python3.10/concurrent/futures/process.py", line 575, in _chain_from_iterable_of_lists
    for element in iterable:
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 621, in result_iterator
    yield _result_or_cancel(fs.pop())
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 319, in _result_or_cancel
    return fut.result(timeout)
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 458, in result
    return self.__get_result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
ValueError: X has 258 features, but TruncatedSVD is expecting 60 features as input.

```

samuelgarcia, 2023-11-28:
Hi thanks for feedback.
I have a branch that fix that.
I will open a PR very soon.

samuelgarcia, 2023-11-28:
Oups. No I think this is already fixed in the main. Can you try to update with the main branch ?

rat-h, 2023-11-28:
`pip install git+https://github.com/SpikeInterface/spikeinterface` solves the problem. Thank you.