SpikeInterface/spikeinterface #2272: Kilosort: 'Error using indexing'

pantelisantonoudiou, 2023-11-29:
Hello, I get an indexing error that originates from MATLAB when I try to run kilosort3 with our data. I tested our data with other sorters such as pykilosort, spykingcircus and mountainsort5 and they get sorted fine and waveforms can be  extracted and loaded to the sortingview gui. This indexing error does not show up when I run kilosort3 with the toy example even when using the same sampling frequency, number of channels, probe settings, and data type (float 32), so I am not sure what's causing the error.

- Here my code.

```Python3
  # Load the recording
  raw_recording = BinaryRecordingExtractor.load_from_folder('raw_data')

  # Run spike sorting
  sorter = 'kilosort3'
  sorting = ss.run_sorter(sorter_name=sorter,
                  recording=raw_recording,
                  output_folder=sorter + "_sorting",
                  verbose=True,
                  docker_image=True,)
  print(sorting)
````

- Here is the  error traceback.

```Python3
"runtime_trace": [
        "Warning: X does not support locale C.UTF-8",
        "Time   0s. Computing whitening matrix..",
        "Getting channel whitening matrix...",
        "Channel-whitening matrix computed.",
        "Time   2s. Loading raw data and applying filters...",
        "Time   2s. Finished preprocessing 16 batches.",
        "Drift correction ENABLED",
        "----------------------------------------Error using indexing",
        "Subscript indices must either be real positive integers or logicals."
    ],
````

I can reproduce this error when Drift correction is DISABLED and also when using kilosort2.5.

- Traceback when drift correction is disabled.

```Python3
    "runtime_trace": [
        "Warning: X does not support locale C.UTF-8",
        "Time   0s. Computing whitening matrix..",
        "Getting channel whitening matrix...",
        "Channel-whitening matrix computed.",
        "Time   2s. Loading raw data and applying filters...",
        "Time   2s. Finished preprocessing 16 batches.",
        "Drift correction DISABLED",
        "----------------------------------------Error using indexing",
        "Subscript indices must either be real positive integers or logicals."
    ],
````

-----------------------------------------------

- Associated json files of the recording.
[si_folder.json](https://github.com/SpikeInterface/spikeinterface/files/13501057/si_folder.json)
[binary.json](https://github.com/SpikeInterface/spikeinterface/files/13501058/binary.json)
[probe.json](https://github.com/SpikeInterface/spikeinterface/files/13501059/probe.json)
[provenance.json](https://github.com/SpikeInterface/spikeinterface/files/13501060/provenance.json)

- My conda environment.
[env.txt](https://github.com/SpikeInterface/spikeinterface/files/13501070/env.txt)
-----------------------------------------------

Thank you for your help!

pantelisantonoudiou, 2023-11-29:
Thank you for your prompt response! I will open an issue on their page.

alejoe91, 2023-12-04:
clogin this one then :)