SpikeInterface/spikeinterface #2274: Feature Request : add waveform.select_units() support when merged unit ids

Antho2422, 2023-11-29:
Hi all, 

I just encountered an issue coming from the method of the WaveformExtractor :
```python
wf.select_units()
``` 

I get the following error : 

```python
AssertionError: unit ids are not all in parents
``` 

If I'm not mistaken, I think this is because the units ids changed after a curation using Phy. In my script I have one WaveformExtractor object containing several units that I've been curated and merged. I would like to create a new WaveformsExtractor object with the new waveforms from the sorting that has just been loaded from Phy.  The units ids changed and the old WaveformsExtractor object does not know the new unit_ids. 

Is it possible to handle this case within the select_units function or do I have to re-extract the waveforms from the new sorting object ? 

Thank you, 
Anthony

alejoe91, 2023-11-30:
@Antho2422 currently this is not supported and you would need to re-extract waveforms. The main complication is merges, since both waveforms / templates and quality metrics would need to be recomputed. 

If you exported to Phy in SpikeInterface and you only removed `noise` clusters, then the unit ids should correspond. Can you share a more complete script that is failing? How are you computing the waveforms and selecting units?

Antho2422, 2024-01-10:
Hi @alejoe91, 

Sorry for the late response but I reextracted the waveforms and it works perfectly. I did merges so of course it was not possible to select units from the non-curated sorting.

Thank you

alejoe91, 2024-01-10:
Glad that it works! I'll close the issue then

closed 2024-01-10 by alejoe91