SpikeInterface/spikeinterface #2278: `plot_timeseries`  time axis is incorrect if requesting time longer than the recording.

JoeZiminski, 2023-11-30:
This is only a small problem and would be fixed with #1989, but just logging it here. If `plot_timeseries` is called with a `time_range` that is larger than that available in the file, it will be plot with an incorrect time axis.

For example, calling `plot_timeseries` with `time_range=50` on a 1-second recording has the take axis range `[0, 50]`. It would be better to throw an error or only show the real time values.

zm711, 2023-11-30:
@JoeZiminski , out of curiosity which version are you using. `plot_timeseries` doesn't exist any more. It has been changed to `plot_traces` :)

https://github.com/SpikeInterface/spikeinterface/blob/78761bc811cfc424a6a20a67dfbaaeafb5a2f176/src/spikeinterface/widgets/traces.py#L119C1-L123

That is the code that we should fix with something like this maybe:

```python
fs = rec0.get_sampling_frequency()
if time_range is None:
    time_range = (0, 1.0)
time_range = np.array(time_range)
assert time_range[1] < rec0.get_duration(segment_index=segment_index), (f'max time range should be '
                                                          f'{rec0.get_duration(segment_index=segment_index}')
```

Although in the actual traces retrieval portion of the code it just silently clips to the correct length. Thus the only real problem is that xlim that is being fed into the graph. So the other option would be something like this (which then also warns them that we are clipping the data retrieval as well):

```python
fs = rec0.get_sampling_frequency()
if time_range is None:
    time_range = (0, 1.0)
time_range = np.array(time_range)
if time_range[1] > rec0.get_duration(segment_index=segment_index):
    warn.warnings("you've enter a time value out of range, range will be clipped accordingly")
    time_range[1] = rec0.get_duration(segment_index=segment_index)
```

I think I would just prefer to be warned that I messed up and let the plotting fix itself rather than break my flow with an assert, but maybe that's the wrong approach...

zm711, 2023-12-01:
Cool I'll put in a PR for this as long as @alejoe91 and @samuelgarcia are okay with the warning/clip approach and do not prefer raising an error?

alejoe91, 2023-12-01:
Yeo sounds good for me!

PR #2286 (2023-12-04): Prevent users from providing a time_range after the ending of the segment in `plot_traces`
Fixes #2278.

Do we want to also add in either an assert or clipping mechanism for negative values for time range while we are editing it?
closed 2023-12-04 by alejoe91