SpikeInterface/spikeinterface #2280: Adding a warning or check for string inputs in ids_to_indices

LeMuellerGuy, 2023-12-01:
Hello everyone,

Since ids are (mostly?) strings and `ids_to_indices` does not produce an error when entering strings because they are in fact iterable, I think it would be helpful for beginner users to at least add a warning if not an outright error to the `ids_to_indices` function to prevent unwanted behaviour. See the following example:

```
recording = se.MaxwellRecordingExtractor(local_path, stream_name="well000", rec_name="rec0000", install_maxwell_plugin=False)
# produces a faulty input, in this case the index to id 401 is 448, which actually crashes h5py
# ids_to_indices outputs [4,4,8] and not [448] because the string input is split into individual chars before being converted to int
trw = plot_traces(recording, channel_ids=recording.channel_ids[401], mode = "auto", backend="matplotlib", show_channel_ids=True)

# this works fine since the single string is encapsulated in a list
trw = plot_traces(recording, channel_ids=[recording.channel_ids[401]], mode = "auto", backend="matplotlib", show_channel_ids=True)
```

A possible comfort increase for this could look like this (added in the `ids_to_indices` function):
`if isinstance(ids, str): print("string channel ids may produce undesireable output")`
or even
`assert isinstance(ids, str), "string channel ids must be encapsulated in a list or array"`

I have already looked into just encapsulating the string for the user, however this raises further issues with other functions (in the example's case the plotting)

h-mayorquin, 2023-12-04:
Agree with @alejoe91 that the input should be strict about the input being a list. I think that the current typing is too permisive and the lack of documentation of the function does not help either.

h-mayorquin, 2023-12-13:
Was solved by this:
https://github.com/SpikeInterface/spikeinterface/pull/2301

closed 2023-12-13 by h-mayorquin