SpikeInterface/spikeinterface #2296: Add `ruff` linting.

JoeZiminski, 2023-12-05:
[Ruff](https://github.com/astral-sh/ruff) is a linter that essentially combines isort and flake8 (written in rust so it is extremely fast). Further to #1703 #1349 I would suggest adding Ruff to the linting pipeline.

I would say this linting is very nice beause you will almost immediately see tangible benefits from using it day-to-day (i.e. flagging unused variables, undefined variables caused by refactoring). It really gives nice sense of security when working. 

Typing is even better, but that's a conversation for another day 😄

zm711, 2023-12-06:
I was checking the Ruff issue tracker and it seems they are still working out some of the auto-fixes. The linter has some bugs with complicated logic where it converts the logic to the wrong value. 

I know Ruff lets you choose which fixes you want to follow with even more customizability than Black (and the variable problem does occur in this code base), but have you run into any issues with the bad autocorrect or do you use it just more as a drop in replacement for black/with black without the autofix?

I know one concern with isort was the potential for circular imports (not sure if that has been alleviated).

JoeZiminski, 2023-12-06:
Thanks we have never had a problem with the autofix across quite a few repos, that been said I can't vouch that it will not cause problems across the whole spikeinterface codebase. We use it alongside black, although it's use is in part as a formatter (e.g. imports) it also performs some checks on the actual code logic (e.g. missing variables).

I think a good first step is that I could open a PR and see what kind of errors it raises, and all existing errors should be fixed manually to avoid any issues with the `--fix` argument. Any errors we don't agree with can be ignored (as you mention it is [extremely flexible](https://docs.astral.sh/ruff/rules/)). 

Some problems may come from it trying to fix things that it thinks are issues but are actually intended.  For the most part, I think this will highlight things that should be fixed e.g. imports not at the top of file used to avoid circular import errors can be solved with a refactoring (though, easier said that done!).

zm711, 2023-12-08:
Looks to me like a lot are from testing, which makes sense since those are tweaked and it would be easy for people to forget to clean up imports. 

I believe something like `preprocessinglist` needs those unused imports because it is the container for allowing the general si import structure, so those errors are (I think), wrong for the spikeinterface code base.

zm711, 2024-06-19:
@JoeZiminski, I will close this for now since I don't think we will get Sam on board. And we have the black linting.

closed 2024-06-19 by zm711

samuelgarcia, 2024-06-20:
Feel free to re open around christmas.