SpikeInterface/spikeinterface #2306: Loading in Blackrock Data

mtjacobsResearcher, 2023-12-06:
My lab is very new to using Spike Interface. We are trying to load Blackrock data in, but we are getting an error. 

AssertionError: Inconsistent ns2 and nev file. 559 segments present in .nev file, but 280 in ns2 file.

It seems to be always twice as many seggments in the .nev than in the .ns2 or .ns5 file. Can you provide any guidance on why we might be facing this issue?

mtjacobsResearcher, 2023-12-08:
Sounds good. I will show you everything I have going on. Basically we took the tutorial Jupiter notebook, and tried to make our own cells doing what the tutorial did but with our own data. We got stuck on step one. And no, the data sets are extremely tiny. I am currently testing on single cell recordings, but we plan on making a pipeline for utah arrays afterwards. 

#Matt & Adarsh's test 
base_folder2=Path('\\Users\\User\Documents\\GitHub\Spike interface tutorial\\')
oe_folder2 = base_folder2 / 'BR_test_2/Tony_2023_12_04_001.ns5'
print(oe_folder2)
full_raw_rec2 = si.read_blackrock(oe_folder2)

#this code works when we only had a .ns5 in the folder. I have tried it for several different days of recordings and the .nev always have 2*n+1 segments to the n segments of the .nsx (2 or 5).

Here is the output when I add a .nev
{
	"name": "AssertionError",
	"message": "Inconsistent ns5 and nev file. 773 segments present in .nev file, but 387 in ns5 file.",
	"stack": "---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
c:\\Users\\User\\Documents\\GitHub\\Spike interface tutorial\\SpikeInterface_Tutorial.ipynb Cell 11 line 5
      <a href='vscode-notebook-cell:/c%3A/Users/User/Documents/GitHub/Spike%20interface%20tutorial/SpikeInterface_Tutorial.ipynb#Y442sZmlsZQ%3D%3D?line=2'>3</a> oe_folder2 = base_folder2 / 'BR_test_2/Tony_2023_12_04_001.ns5'
      <a href='vscode-notebook-cell:/c%3A/Users/User/Documents/GitHub/Spike%20interface%20tutorial/SpikeInterface_Tutorial.ipynb#Y442sZmlsZQ%3D%3D?line=3'>4</a> print(oe_folder2)
----> <a href='vscode-notebook-cell:/c%3A/Users/User/Documents/GitHub/Spike%20interface%20tutorial/SpikeInterface_Tutorial.ipynb#Y442sZmlsZQ%3D%3D?line=4'>5</a> full_raw_rec2 = si.read_blackrock(oe_folder2)

File c:\\Users\\User\\miniconda3\\envs\\si_env\\lib\\site-packages\\spikeinterface\\extractors\
eoextractors\\blackrock.py:51, in BlackrockRecordingExtractor.__init__(self, file_path, stream_id, stream_name, block_index, all_annotations, use_names_as_ids)
     49 if \".ns\" in suffix:
     50     neo_kwargs[\"nsx_to_load\"] = int(suffix[-1])
---> 51 NeoBaseRecordingExtractor.__init__(
     52     self,
     53     stream_id=stream_id,
     54     stream_name=stream_name,
     55     all_annotations=all_annotations,
     56     use_names_as_ids=use_names_as_ids,
     57     **neo_kwargs,
     58 )
     59 self._kwargs.update({\"file_path\": str(Path(file_path).absolute())})

File c:\\Users\\User\\miniconda3\\envs\\si_env\\lib\\site-packages\\spikeinterface\\extractors\
eoextractors\
eobaseextractor.py:185, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    156 def __init__(
    157     self,
    158     stream_id: Optional[str] = None,
   (...)
    163     **neo_kwargs: Dict[str, Any],
    164 ) -> None:
    165     \"\"\"
    166     Initialize a NeoBaseRecordingExtractor instance.
    167 
   (...)
    182 
    183     \"\"\"
--> 185     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
    187     kwargs = dict(all_annotations=all_annotations)
    188     if block_index is not None:

File c:\\Users\\User\\miniconda3\\envs\\si_env\\lib\\site-packages\\spikeinterface\\extractors\
eoextractors\
eobaseextractor.py:25, in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     23 def __init__(self, block_index, **neo_kwargs):
     24     if not hasattr(self, \"neo_reader\"):  # Avoid double initialization
---> 25         self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
     27     if self.neo_reader.block_count() > 1 and block_index is None:
     28         raise Exception(
     29             \"This dataset is multi-block. Spikeinterface can load one block at a time. \"
     30             \"Use 'block_index' to select the block to be loaded.\"
     31         )

File c:\\Users\\User\\miniconda3\\envs\\si_env\\lib\\site-packages\\spikeinterface\\extractors\
eoextractors\
eobaseextractor.py:64, in _NeoBaseExtractor.get_neo_io_reader(cls, raw_class, **neo_kwargs)
     62 neoIOclass = getattr(rawio_module, raw_class)
[... 13 lines cut ...]
--> 179     self._parse_header()
    180     self._check_stream_signal_channel_characteristics()

File c:\\Users\\User\\miniconda3\\envs\\si_env\\lib\\site-packages\
eo\\rawio\\blackrockrawio.py:359, in BlackrockRawIO._parse_header(self)
    357 if self._avail_files['nev']:
    358     for nsx_nb in self.nsx_to_load:
--> 359         self.__match_nsx_and_nev_segment_ids(nsx_nb)
    361 self.nsx_datas = {}
    362 self.sig_sampling_rates = {}

File c:\\Users\\User\\miniconda3\\envs\\si_env\\lib\\site-packages\
eo\\rawio\\blackrockrawio.py:1269, in BlackrockRawIO.__match_nsx_and_nev_segment_ids(self, nsx_nb)
   1266             ev_ids[mask_after_seg] += 1
   1268 # consistency check: same number of segments for nsx and nev data
-> 1269 assert nb_possible_nev_segments == len(nonempty_nsx_segments), \\
   1270     ('Inconsistent ns{0} and nev file. {1} segments present in .nev file, but {2} in '
   1271      'ns{0} file.'.format(nsx_nb, nb_possible_nev_segments,
   1272                           len(nonempty_nsx_segments)))
   1274 new_nev_segment_id_mapping = dict(zip(range(nb_possible_nev_segments),
   1275                                       sorted(list(nonempty_nsx_segments))))
   1277 # replacing event ids by matched event ids in place

AssertionError: Inconsistent ns5 and nev file. 773 segments present in .nev file, but 387 in ns5 file."
}

samuelgarcia, 2025-07-02:
yep