SpikeInterface/spikeinterface #2309: Concerns Regarding Unit Measurement in compute_template_metrics Function for Waveform Extractor Objects

Anyesz, 2023-12-07:
My concern arises when applying the function compute_template_metrics from the postprocessing module to waveform objects. According to the documentation, the metrics repolarization_slope and recovery_slope are provided in V/m, but in my case, these values are 6-digit numbers, which seems extremely large. I am questioning whether the unit of measurement is incorrect or if there might be a missing conversion somewhere in the process.

For instance, if we consider the approximate peak-to-trough duration of a realistic cell as 0.5 ms, and the membrane potential changes by 100 mV, the expected repolarization speed should be 100 mV/0.5 ms = 200 V/s. However, the values I am obtaining are around 200,000, which, in my opinion, appears to be in mV/s instead of V/s.

JoeZiminski, 2024-06-20:
@alejoe91 I am confused about this, [e.g. for `repolarization_slope`](https://github.com/SpikeInterface/spikeinterface/blob/328bba007391c47a5f08a1f0897119b5ea09bc05/src/spikeinterface/postprocessing/template_metrics.py#L407) the slope is calculated directly from the template, is the template in V? or this depends on the recording scaling?

chrishalcrow, 2024-06-24:
Hello, I had a go at updated the docs in response to this issue.

@Anyesz I think the correct unit are (unit of your template)/seconds. The unit of your template will depend on your recording: our lab is using Neuropixels 2.0 and the open ephys format. This uses the `bit_volts` parameter to convert from the integer raw data to "unitful data". For us this number is 0.195 and I'm pretty sure it converts integers to mV. So our templates are in mV, and we get similar results to you. Maybe you could check to see if your templates are in mV too, which would explain why your values are around 200,000.

alejoe91, 2024-06-24:
@chrishalcrow 0.195 is the conversion factor to uV ;)

chrishalcrow, 2024-07-19:
Hello, the docs have been updated (https://github.com/SpikeInterface/spikeinterface/pull/3066) so I'm closing this. @Anyesz if you think it's not clear feel free to re-open.

closed 2024-07-19 by chrishalcrow