SpikeInterface/spikeinterface #2310: Some Problems about  BrokenProcessPool in SpikeTutorials 0.99

holawa, 2023-12-07:
Dear developers !
I am using 0.99.1 SpikeInterface on windows10, my computer has 64GB RAM, and I meet some problems when I try to learn something from the tutorial. BrokenProcessPool seems to occur everywhere. Indeed I doubt it occurs when I try to use the Variable **job_kwargs**,wether in **.save()** , **spykingcircus2** or **tridesclous2** 
Actually, serveral weeks ago, I didn't meet BrokenProcessPool when I followed the tutorial. And I'm really need your help.
### ①In the [**Official_Tutorial_SI_0.99_Nov23**] **Capter 3** **Saving and loading SpikeInterface objects**
```
if (base_folder / "preprocessed_compressed.zarr").is_dir():
    recording_saved = si.read_zarr(base_folder / "preprocessed_compressed.zarr")
else:
    import numcodecs
    compressor = numcodecs.Blosc(cname="zstd", clevel=9, shuffle=numcodecs.Blosc.BITSHUFFLE)
    recording_saved = recording_sub.save(format="zarr", folder=base_folder / "preprocessed_compressed.zarr",
                                         compressor=compressor,
                                         **job_kwargs)
```
When I run the code block above, it returns BrokenProcessPool information
```
BrokenProcessPool                         Traceback (most recent call last)
Cell In[57], line 6
      4 import numcodecs
      5 compressor = numcodecs.Blosc(cname="zstd", clevel=9, shuffle=numcodecs.Blosc.BITSHUFFLE)
----> 6 recording_saved = recording_sub.save(format="zarr", folder=base_folder / "preprocessed_compressed.zarr",
      7                                      compressor=compressor,
      8                                      **job_kwargs)

File ~\anaconda3\Lib\site-packages\spikeinterface\core\base.py:777, in BaseExtractor.save(self, **kwargs)
    775     loaded_extractor = self.save_to_memory(**kwargs)
    776 elif format == "zarr":
--> 777     loaded_extractor = self.save_to_zarr(**kwargs)
    778 else:
    779     loaded_extractor = self.save_to_folder(**kwargs)

File ~\anaconda3\Lib\site-packages\spikeinterface\core\base.py:964, in BaseExtractor.save_to_zarr(self, name, folder, storage_options, channel_chunk_size, verbose, zarr_path, **save_kwargs)
    962 save_kwargs["storage_options"] = storage_options
    963 save_kwargs["channel_chunk_size"] = channel_chunk_size
--> 964 cached = self._save(verbose=verbose, **save_kwargs)
    966 # save properties
    967 prop_group = zarr_root.create_group("properties")

File ~\anaconda3\Lib\site-packages\spikeinterface\core\baserecording.py:514, in BaseRecording._save(self, format, **save_kwargs)
    508     zarr_kwargs["compressor"] = compressor = get_default_zarr_compressor()
    509     print(
    510         f"Using default zarr compressor: {compressor}. To use a different compressor, use the "
    511         f"'compressor' argument"
    512     )
--> 514 write_traces_to_zarr(self, **zarr_kwargs, **job_kwargs)
    516 # save probe
    517 if self.get_property("contact_vector") is not None:

File ~\anaconda3\Lib\site-packages\spikeinterface\core\core_tools.py:709, in write_traces_to_zarr(recording, zarr_root, zarr_path, storage_options, dataset_paths, channel_chunk_size, dtype, compressor, filters, verbose, auto_cast_uint, **job_kwargs)
    705 init_args = (recording, zarr_path, storage_options, dataset_paths, dtype, cast_unsigned)
    706 executor = ChunkRecordingExecutor(
    707     recording, func, init_func, init_args, verbose=verbose, job_name="write_zarr_recording", **job_kwargs
    708 )
--> 709 executor.run()

File ~\anaconda3\Lib\site-packages\spikeinterface\core\job_tools.py:399, in ChunkRecordingExecutor.run(self)
    396 if self.progress_bar:
    397     results = tqdm(results, desc=self.job_name, total=len(all_chunks))
--> 399 for res in results:
    400     if self.handle_returns:
    401         returns.append(res)

File ~\anaconda3\Lib\site-packages\tqdm\notebook.py:254, in tqdm_notebook.__iter__(self)
    252 try:
    253     it = super(tqdm_notebook, self).__iter__()
--> 254     for obj in it:
    255         # return super(tqdm...) will not catch exception
    256         yield obj
    257 # NB: except ... [ as ...] breaks IPython async KeyboardInterrupt

[... 194 lines cut ...]
    processor.run()
  File "C:\Users\Melody\anaconda3\Lib\site-packages\spikeinterface\core\job_tools.py", line 399, in run
    for res in results:
  File "C:\Users\Melody\anaconda3\Lib\site-packages\tqdm\notebook.py", line 254, in __iter__
    for obj in it:
  File "C:\Users\Melody\anaconda3\Lib\site-packages\tqdm\std.py", line 1178, in __iter__
    for obj in iterable:
  File "C:\Users\Melody\anaconda3\Lib\concurrent\futures\process.py", line 606, in _chain_from_iterable_of_lists
    for element in iterable:
  File "C:\Users\Melody\anaconda3\Lib\concurrent\futures\_base.py", line 619, in result_iterator
    yield _result_or_cancel(fs.pop())
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Melody\anaconda3\Lib\concurrent\futures\_base.py", line 317, in _result_or_cancel
    return fut.result(timeout)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Melody\anaconda3\Lib\concurrent\futures\_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Melody\anaconda3\Lib\concurrent\futures\_base.py", line 401, in __get_result
    raise self._exception
concurrent.futures.process.BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.

Spike sorting failed. You can inspect the runtime trace in C:\Users\Melody\Desktop\test1\results_TC22/spikeinterface_log.json.
```
If I run the sorters in Docker, such as kilosort2, mountainsort4, tridecircus, ironclust, etc. They're all good, except when I run a huge(maybe) data in spykingsort image(the tutorial data is OK with no error), I meet the same problem with [Error while running dockerized SpykingCircus #895](https://github.com/SpikeInterface/spikeinterface/issues/895).However,I feel sorry that I am just a new learner , I can't manage to solve the problem by setting the parameters (just because I don't know where to set , use whith function to set).

samuelgarcia, 2023-12-08:
Hi @holawa 

1.
for this kind of bug that occur in multiprocessing (n_jobs>1)
> BrokenProcessPool: A process in the process pool was terminated abruptly while the future was running or pending.

The easiest is to use n_jobs=1 and get a better trace for the bug.

2. We did a recent fix spkykingcircus in code if you install spikeinterface from source maybe this could fix it.

3. Same as 1

In short could try to run the same notebook with n_jobs=1 ?
And repost error here ?

samuelgarcia, 2023-12-08:
For 1, this is a good news but this must be very slow. Maybe you could try a few n_jobs.
How many core do you have  on your machine ?
For the bug 2 and 3 I think, this is also because inetrnally tdc2 orsc2 are doing multiporcessing and sometimes is going wrong in your case.

zm711, 2023-12-08:
I haven't directly tried the dataset from the tutorial, but using some simulated data, and mimicking the tutorial, I am able to get multiprocessing to work. I think @h-mayorquin's [comment](https://github.com/SpikeInterface/spikeinterface/issues/2122#issuecomment-1803723629), may be pertinent to this in that on Windows there is an inconsistent need to use name==main in Windows specifically in the Spikeinterface context. I never need to use (thus far), but I also don't try to run full scripts (I typically do a more cell by cell approach).

Other possibilities I can think of is that my computer isn't hyperthreaded so it has 8 physical and 8 logical cores. I'm not sure if the multiprocessing for Windows struggles with logical cores or not.
I have administrator privileges, but I don't typically run my python as admin, so I don't think that should be the problem.

@holawa,

One question I have is how are you running the tutorial. In a juypter notebook or did you try to copy the cells into a standard `.py` file? I'm not sure how `if __name__ == '__main__'` could fit into a juypter context, but maybe @samuelgarcia could answer that better.

zm711, 2023-12-08:
Could you also let us know the version of threadpoolctl you have? You can just type `conda list` in the environment containing spikeinterface. (Although it appears you are working in base instead of a conda env is that true?)

holawa, 2023-12-08:
I feel sorry that I failed.
I wrote a code block like this, and replaced the old one
<pre><code>
job_kwargs = dict(n_jobs=4, chunk_duration='1s', progress_bar=True)
if __name__ == '__main__':
    if (base_folder / "preprocessed_compressed.zarr").is_dir():
        recording_saved = si.read_zarr(base_folder / "preprocessed_compressed.zarr")
    else:
        compressor = numcodecs.Blosc(cname="zstd", clevel=9, shuffle=numcodecs.Blosc.BITSHUFFLE)
        recording_saved = recording_sub.save(format="zarr", folder=base_folder / "preprocessed_compressed.zarr",
                                         compressor=compressor,
                                         **job_kwargs)
</pre></code>
The code returned
<details>
 <summary><code>Click to expand!</code></summary>
<pre><code>
write_zarr_recording with n_jobs = 4 and chunk_size = 30000
write_zarr_recording: 0%
0/300 [00:01<?, ?it/s]
---------------------------------------------------------------------------
BrokenProcessPool                         Traceback (most recent call last)
Cell In[50], line 7
      5 else:
      6     compressor = numcodecs.Blosc(cname="zstd", clevel=9, shuffle=numcodecs.Blosc.BITSHUFFLE)
----> 7     recording_saved = recording_sub.save(format="zarr", folder=base_folder / "preprocessed_compressed.zarr",
      8                                      compressor=compressor,
      9                                      **job_kwargs)

File ~\Downloads\spikeinterface_main\src\spikeinterface\core\base.py:828, in BaseExtractor.save(self, **kwargs)
    826     loaded_extractor = self.save_to_memory(**kwargs)
    827 elif format == "zarr":
--> 828     loaded_extractor = self.save_to_zarr(**kwargs)
    829 else:
    830     loaded_extractor = self.save_to_folder(**kwargs)

File ~\Downloads\spikeinterface_main\src\spikeinterface\core\base.py:1015, in BaseExtractor.save_to_zarr(self, name, folder, storage_options, channel_chunk_size, verbose, zarr_path, **save_kwargs)
   1013 save_kwargs["storage_options"] = storage_options
   1014 save_kwargs["channel_chunk_size"] = channel_chunk_size
-> 1015 cached = self._save(verbose=verbose, **save_kwargs)
   1017 # save properties
   1018 prop_group = zarr_root.create_group("properties")

File ~\Downloads\spikeinterface_main\src\spikeinterface\core\baserecording.py:519, in BaseRecording._save(self, format, **save_kwargs)
    513     zarr_kwargs["compressor"] = compressor = get_default_zarr_compressor()
    514     print(
    515         f"Using default zarr compressor: {compressor}. To use a different compressor, use the "
    516         f"'compressor' argument"
    517     )
--> 519 write_traces_to_zarr(self, **zarr_kwargs, **job_kwargs)
    521 # save probe
    522 if self.get_property("contact_vector") is not None:

File ~\Downloads\spikeinterface_main\src\spikeinterface\core\core_tools.py:709, in write_traces_to_zarr(recording, zarr_root, zarr_path, storage_options, dataset_paths, channel_chunk_size, dtype, compressor, filters, verbose, auto_cast_uint, **job_kwargs)
    705 init_args = (recording, zarr_path, storage_options, dataset_paths, dtype, cast_unsigned)
    706 executor = ChunkRecordingExecutor(
    707     recording, func, init_func, init_args, verbose=verbose, job_name="write_zarr_recording", **job_kwargs
    708 )
--> 709 executor.run()

File ~\Downloads\spikeinterface_main\src\spikeinterface\core\job_tools.py:379, in ChunkRecordingExecutor.run(self)
    376 if self.progress_bar:
    377     results = tqdm(results, desc=self.job_name, total=len(all_chunks))
--> 379 for res in results:
    380     if self.handle_returns:
    381         returns.append(res)

File ~\anaconda3\Lib\site-packages\tqdm\notebook.py:254, in tqdm_notebook.__iter__(self)
    252 try:
    253     it = super(tqdm_notebook, self).__iter__()
[... 1032 lines cut ...]
webcolors                         1.13
webencodings                      0.5.1
websocket-client                  0.58.0
Werkzeug                          2.2.3
whatthepatch                      1.0.2
wheel                             0.38.4
widgetsnbextension                4.0.5
win-inet-pton                     1.1.0
wrapt                             1.14.1
xarray                            2023.6.0
xlwings                           0.29.1
xxhash                            2.0.2
xyzservices                       2022.9.0
yapf                              0.31.0
yarl                              1.9.3
zarr                              2.16.1
zict                              3.0.0
zipp                              3.11.0
zope.interface                    5.4.0
zstandard                         0.19.0
PS C:\Users\Melody\Desktop\test0> 
</code></pre> 
</details>

Anyway, thank you! @zm711

holawa, 2023-12-09:
Really, really, really thank you! @zm711 
I update the `threadpoolctl 2.2 `to `threadpoolctl 3.2` and ① ② ③ went all well. Even if I set the `n_jobs` to other numbers, the code went stable and I fell much better !

> No worries. There is a known bug in threadpoolctl 2.x, so it might be worth while to update threadpoolctl to 3+ and see if that fixes it. The threadpoolctl 2.x can lead to the `None` not having attribute `split`. Could you try to update and then repeat the sorting and see if that helps?

h-mayorquin, 2023-12-09:
Thanks @zm711. Maybe we should bump up the minimal version of threadpoolctl. I will take a look at that and then close this.

zm711, 2023-12-09:
Maybe with the switch in #2218 it led to this issue with threadpoolctl coming up. It happened two weeks ago so the timeline makes sense. But yeah based on my reading I think it is probably better just to have a minimal version (but if you discover something else with your reading do let me know!)

alejoe91, 2023-12-12:
Not sure it's related to #2218 since that only affected the PCA computation and quality metrics

zm711, 2023-12-12:
You’re right. I think it was just an old install. I support @h-mayorquin ‘s idea to just put in a minimum version so that if someone has an old install we make sure SI doesn’t have a buggy threadpoolctl.

vncntprvst, 2024-01-19:
Just a note that this error can occur, even with `threadpoolctl >= 3.2.0`, when not enough memory is available (with respect to chunk size). This is something that happens to me sometimes when debugging code on a remote cluster node, and I did not request as much memory as when running jobs. The issue would not easily be fixed by [`job_kwargs` optimization](https://github.com/SpikeInterface/spikeinterface/issues/2232), because the total memory and number of cores on the node is not the same as the number reserved for the job.

zm711, 2024-01-19:
@vncntprvst Thanks for this. If we get enough votes @samuelgarcia might become magically inspired to work on this :). 

I think you raise a good point though. I think there are two separate problems:

1) have threadpoolctl < 3 leads to the actual issue here. the `None` attribute error would be fixed by fixing threadpoolctl.
2) if the job_kwargs are not tuned properly then the parallelization will break (as you see) and currently the job_kwargs require the user to tune rather than having spikeinterface aid the user in tuning them to appropriate conditions. 

I think we need to both bump the threadpoolctl, but also have a job_kwargs_optimizer (but that's my opinion of course). 

Thanks again.

PR #2413 (2024-01-22): Add version limit to `threadpoolctl`
See associated issue #2412, but base Anaconda ships with 2.2.0. 

The short story: < 3 can have some issues that can break parallel processing for users (see https://github.com/SpikeInterface/spikeinterface/issues/2401 and https://github.com/SpikeInterface/spikeinterface/issues/2310).

This PR just sets that limit at 3.0.0.

(Fixes #2412 , Fixes #2401, Fixes #2310)