SpikeInterface/spikeinterface #2332: Issue about read Intan RHD data

sunjie454, 2023-12-14:
When using the read_intan() function in spikeinterface to read Intan data, I can only obtain channel data and not digital-in data. Does spikeinterface provide methods and functions to obtain digital-in data? Also, when Intan data lacks a data block, the data reading fails. Is there a solution for this？

zm711, 2023-12-14:
@sunjie454, so spikeinterface doesn't currently have a utility function to get the digital-in data from intan `rhd` data. But Neo does (the `read_intan` function relies on Neo under the hood). This functionality was added since the last Neo release so you would need to install Neo from source (let me know if you don't know how to do that). You'll likely need to read some of the [neo documentation](https://neo.readthedocs.io/en/latest/) but currently the digital data can be accessed using the `get_analogsignal_chunk` function which will return the raw digital-in data. Then you'll need to convert it based on the [intan documentation](https://intantech.com/downloads.html?tabSelect=Manuals). This involves some bit based operations (again I can help more if you need it once you read the docs).

I'm not sure what you mean exactly by data block? Could you post the exact error you're getting?

zm711, 2023-12-14:
Actually one way you could also access the data through spikeinterface would be like so: 

```python
dig_data_extracted = read_intan(file_path = 'path/to/data', stream_id = '4')
# then bring it into memory
raw_dig_data = dig_data_extracted.get_traces()
# you still need to convert the raw_dig_data into actual dig data with Intan documentation
```

You would still need neo built from source however.

sunjie454, 2024-01-04:
> @sunjie454, so spikeinterface doesn't currently have a utility function to get the digital-in data from intan `rhd` data. But Neo does (the `read_intan` function relies on Neo under the hood). This [...]

Thanks for your answer!
About issue of data block is that the data collected by Intan may become unreadable due to certain reasons such as an automatic shutdown during the data collection process, possibly caused by factors like power loss. 
The error as follows.
[image]

zm711, 2024-01-04:
Unfortunately we can't help with that specific type of data loss. Both Neo (and thus SpikeInterface) rely on generating a memory map of the data so allow for lazy loading. If your file itself is corrupted (due to power loss etc), then the memory map can't be generated and the code will fail. Intan has their own code for loading data (but it does it eagerly, ie everything will go straight into memory so you need enough RAM), you could try their code, but if that fails too then I would say the file is like too corrupted to salvage. You'd have to reach out to Intan and see if they have any recs for recovering part of the data from the file.

sunjie454, 2024-01-08:
> Unfortunately we can't help with that specific type of data loss. Both Neo (and thus SpikeInterface) rely on generating a memory map of the data so allow for lazy loading. If your file itself is [...]

Thank you for your response. Your help has been significant, and I will try using the methods provided by Intan to read the data.

zm711, 2024-01-11:
@sunjie454, I'll close this for now, but if you have any other questions feel free to open new issues as they come up.

jakeswann1, 2024-10-31:
Hi @zm711, I'm also looking to load digital inputs from an Intan .rhd recording, and tried the method you suggested [here](https://github.com/SpikeInterface/spikeinterface/issues/2332#issuecomment-1855914238). I have 8 digital lines enabled in this particular recording, and am using SI v0.101.2. Using the [example code Intan provides for loading](https://intantech.com/files/load_intan_rhd_format.zip), I can visualise the correct signal on all the lines, but when I try and load them in neo/spikeinterface, only the first line is loaded. Any ideas why this might be?
[image]
[image]

zm711, 2024-10-31:
@jakeswann1 we recently updating the structure for neo so this should work automatically now. You would need the most recently Neo ( I forget if we added it in 0.13.2 or 0.13.3). Heberto added in automatic demultiplexing. Which version are you using?

jakeswann1, 2024-10-31:
I was using 0.13.1. Solved, thank you!