SpikeInterface/spikeinterface #2335: Creation of a SharedMemoryRecording

yger, 2023-12-14:
Actually, it would be great to have such a SharedMemoryRecording object, similarly to what has been done with SharedMemorySorting. With such an object, we could preload in memory at the beginning of a spike-sorting pipeline the entire traces (assuming memory is large enough), and speed up drastically internal sorters. Not everyone has recordings larger than 32Go, so it could be useful, especially for speeding up benchmarks and the generate_recording() traces generated on-the-fly. We could pregenerate everything once, and then quickly access the data

samuelgarcia, 2023-12-14:
merci beaucoup camarade de rapeller ce que j'ai promis.

yger, 2023-12-14:
Oui, spécial dédicace :-)

h-mayorquin, 2023-12-15:
We can use the latest pickle protocol out-of-band buffer to pickle extractors containing no memory maps with shared memory (when memory allows):

https://peps.python.org/pep-0574/

This way, we eliminate the distinction of memory vs memmap from the API which is an implementation detail of our ad-hoc parallelization sheme.

zm711, 2024-03-14:
Closed by #2365.

closed 2024-03-14 by zm711