SpikeInterface/spikeinterface #2337: `NwbRecordingExtractor` reads incorrect subset of channels.

TheChymera, 2023-12-14:
I am dealing with an NWB file produced from SpikeGLX NeuroPixels data via the following code:

```
from datetime import datetime
from dateutil import tz
from pathlib import Path
from neuroconv.converters import SpikeGLXConverterPipe
import json

dir_path = "/mnt/DATA/data/studies/manish/mvmnda/sourcedata/sub-M388/M388-2023-11-20_2_g0"
converter = SpikeGLXConverterPipe(folder_path=dir_path)

# Extract what metadata we can from the source files
metadata = converter.get_metadata()

session_start_time = metadata["NWBFile"]["session_start_time"].replace(tzinfo=tz.gettz("US/Pacific"))
metadata["NWBFile"].update(session_start_time=session_start_time)

# Choose a path for saving the nwb file and run the conversion
nwbfile_path = "/mnt/DATA/data/studies/manish/mvmnda/rawdata/my_spikeglx_session.nwb"
converter.run_conversion(nwbfile_path=nwbfile_path, metadata=metadata)
```

When I try to read this file in spikeinterface I get:

```python
>>> import spikeinterface.full as si
>>> myfile="my_spikeglx_session.nwb"
>>> print(si.NwbRecordingExtractor(myfile, electrical_series_name="ElectricalSeriesNIDQ").channel_ids[:20])
['XA0' 'XA1' 'XA2' 'XA3' 'XA4' 'XA5' 'XA6' 'XA7']
>>> print(si.NwbRecordingExtractor(myfile, electrical_series_name="ElectricalSeriesAP").channel_ids[:20])
['XA0' 'XA1' 'XA2' 'XA3' 'XA4' 'XA5' 'XA6' 'XA7' 'AP0' 'AP1' 'AP2' 'AP3'
 'AP4' 'AP5' 'AP6' 'AP7' 'AP8' 'AP9' 'AP10' 'AP11']
```

As I understand it the `XA` channels should be in `NIDQ` but not in `AP`. Is this an actual issue I'm seeing here? If so any idea whether it's on the file creation or file reading side, and what I might do to fix it?

ATTN: @manimoh

manimoh, 2023-12-14:
Looks like the blame might lie in these [lines](https://github.com/SpikeInterface/spikeinterface/blob/b7b891bd63d68ce8dd5321286c92ef6086db7af2/src/spikeinterface/extractors/nwbextractors.py#L266-L273) 

The `electrical_series.electrodes["channel_name"][:]s` is contains the same stream with all channels concatenated (NIDQ analog lines first, and the AP channels from the probe next), and then it just reads the first 8 or the first 384 indices based on the channel count according to the parameter `electrical_series_name` passed onto the `NWBRecordingExtractor()`

alejoe91, 2023-12-14:
@h-mayorquin could you take a look? Thanks @manimoh, I agree that we should slice the channel names based on the electrodes column!

h-mayorquin, 2023-12-15:
>  The electrical_series.electrodes["channel_name"][:]s is contains the same stream with all channels concatenated (NIDQ analog lines first, and the AP channels from the probe next

This should not be. This is a problem in the file creation side, neuroconv:

```python
from datetime import datetime
from dateutil import tz
from pathlib import Path
from neuroconv.converters import SpikeGLXConverterPipe

ECEPHY_DATA_PATH = Path("C:\\Users\\hmayo\\neuroconv_test_data\\ephy_testing_data").resolve()
folder_path = Path(f"{ECEPHY_DATA_PATH}/spikeglx/Noise4Sam_g0")
folder_path = folder_path.resolve()
assert folder_path.exists(), f"Folder {folder_path} does not exist"
converter = SpikeGLXConverterPipe(folder_path=folder_path)

# Extract what metadata we can from the source files
metadata = converter.get_metadata()
# For data provenance we add the time zone information to the conversion
session_start_time = metadata["NWBFile"]["session_start_time"].replace(tzinfo=tz.gettz("US/Pacific"))
metadata["NWBFile"].update(session_start_time=session_start_time)

# Choose a path for saving the nwb file and run the conversion
output_folder = Path("./spike_glx_output")
nwbfile_path = output_folder / "my_spikeglx_session.nwb"
nwbfile_path.parent.mkdir(exist_ok=True, parents=True)
converter.run_conversion(nwbfile_path=nwbfile_path, metadata=metadata, overwrite=True)

# Read the nwb file and check that the metadata was saved correctly
from pynwb import NWBHDF5IO
from pynwb import NWBFile

io = NWBHDF5IO(nwbfile_path, mode="r")
nwbfile = io.read()

nwbfile.acquisition["ElectricalSeriesNIDQ"].electrodes.to_dataframe()["channel_name"]
# shoud be    ['XA0', 'XA1', 'XA2', 'XA3', 'XA4', 'XA5', 'XA6', 'XA7
# But outputs :
0    AP0
1    AP1
2    AP2
3    AP3
4    AP4
5    AP5
6    AP6
7    AP7
```

h-mayorquin, 2023-12-15:
Created at issue at neuroconv and should be fixed there:
https://github.com/catalystneuro/neuroconv/issues/687

h-mayorquin, 2023-12-15:
@TheChymera @manimoh 
I opened a PR in neuroconv that fixes this from the writing side. That is, it correctly writes the electrode table region to each of the `ElectricalSeries` objects. What remains to be seen is if we are not making any mistake in this side when reading back from the region.

If you could test the branch at neuroconv that would be great:
https://github.com/catalystneuro/neuroconv/pull/688

h-mayorquin, 2023-12-15:
Actually, I will add the round-trip there at neuroconv. That is a good test for both packages.

h-mayorquin, 2023-12-15:
Added roundtrip at neuroconv side:
https://github.com/catalystneuro/neuroconv/pull/689

It seems that the reading side is working (to my surprise) correctly. Yei! So we did fix this here before as I remembered it seems.

h-mayorquin, 2024-03-20:
OK, I am closing this as I think the fix is clear and we did not get further feedback from OP. Feel free to open another issue.

Related:
https://github.com/SpikeInterface/spikeinterface/issues/2588

closed 2024-03-20 by h-mayorquin