SpikeInterface/spikeinterface #2339: Test failures ( `ModuleNotFoundError: No module named 'spikeinterface.sorters.tests'` ) on `=0.99.1`

TheChymera, 2023-12-14:
I am getting a lot of test failures with `ModuleNotFoundError: No module named 'spikeinterface.sorters.tests'`. Full build log here → https://ppb.chymera.eu/bf16b6.log

Any ideas what might be going on?

zm711, 2023-12-14:
I'm assuming it's because of this:

https://github.com/SpikeInterface/spikeinterface/blob/b7b891bd63d68ce8dd5321286c92ef6086db7af2/pyproject.toml#L30-L45

The tests aren't packaged with spikeinterface. They are only run from source or am I misunderstanding what you are doing?

h-mayorquin, 2023-12-15:
Relevant but probaby different. Linking for provenance:
https://github.com/SpikeInterface/probeinterface/issues/145

zm711, 2023-12-15:
@h-mayorquin, I think that is probably the exact issue, but if there were a solution for probeinterface I would assume the same solution would apply here? Not sure though.

JoeZiminski, 2024-06-20:
@TheChymera thanks for raising this point, @zm711 I think if everyone is happy this can be closed? I agree it is the expected behaviour of `pip` install not to include tests, to run tests the package should be cloned from github. The steps are:

1. git clone <url agress taken from spikeinterface github website>
2. cd to folder and `pip install -e .[dev]` (on macOS this might need to be `pip install -e ".[dev]"`
3. run tests with pytest (e.g. I think `pytest .` should work).

zm711, 2024-06-20:
Yep agree!

closed 2024-06-20 by zm711