SpikeInterface/spikeinterface #2342: Change default argument `durations` argument from list in `generate_recordings.py`

JoeZiminski, 2023-12-15:
Just leaving here so I don't forget, happy to make a PR on this. Some of the default arguments for functions in `generate_recordings.py` are mutable, which is not reccomended due to [pythons crazy behaviour](https://florimond.dev/en/posts/2018/08/python-mutable-defaults-are-the-source-of-all-evil). These could be changed to `None` and the default value set in the body of the function.

h-mayorquin, 2023-12-18:
Yeah, I noticed that too recently. This is a common mistake that people make in Python. We should change it. A good heuristic to remember this is that default values are values (not expressions).

h-mayorquin, 2024-05-16:
This discussion came up again:

https://github.com/SpikeInterface/spikeinterface/pull/2777

 @alejoe91 really wanted to move forward with this, @samuelgarcia supported the idea of using tuples or copying the list at the beginning of the function.

zm711, 2026-10-30:
Where are we will this-- 2 years later?