SpikeInterface/spikeinterface #2344: Excluding a part of the recording accurately

aghatpande, 2023-12-17:
Hi,
I have a large recording object: (ConcatenateSegmentRecording: 64 channels - 32.0kHz - 1 segments - 132,701,374 samples 4,146.92s (1.15 hours) - int16 dtype) and I know samples 79543840 to 82133500 total samples to trim:2589660
 have an artefact due to losing ground temporarily. How do I trim this accurately. I did manage to remove those samples using 
```
data_before_slice = np.array(ordered_raw_rec.get_traces(start_frame=0, end_frame=slice_start))
data_after_slice = np.array(ordered_raw_rec.get_traces(start_frame=slice_end, end_frame=ordered_raw_rec.get_num_frames()))
# Concatenate the segments
trimmed_raw_rec = np.concatenate((data_before_slice, data_after_slice), axis=0)
trimmed_raw_rec = si.numpyextractors.NumpyRecording(trimmed_raw_rec, sampling_frequency=ordered_raw_rec.get_sampling_frequency())
```
This seems to work but I'm not sure that it didn't introduce any artefacts in the recording. Could someone advise how to acieve this by sticking to SI methods rather than switching to the numpy array manipulation. Also saving this object using zarr seems to fail. I would rather use defined SI methods and lazy eval (say by annotating the stretch) so as to ensure reproducibility. Any advice / insight much appreciated.

alejoe91, 2023-12-18:
Hi  @aghatpande 

Indeed you can do what you're trying to do with a few simple SpikeInterface calls. In particular, you'll be manupulating objects as explained here: https://spikeinterface.readthedocs.io/en/latest/modules/core.html#manipulating-objects-slicing-aggregating

Here's how:
```
slice_start = 79543840
slice_end = 82133500
recording_before_slice = ordered_raw_rec.frame_slice(start_frame=0, end_frame=slice_start)
recording_after_slice = ordered_raw_rec.frame_slice(start_frame=slice_end, end_frame=ordered_raw_rec.get_num_samples())

# lazy concatenation of recordings
trimmed_rec = si.concatenate_recordings(recording_list=[recording_before_slice, recording_after_slice])
```

Then you can just continue the analysis with the lazy trimmed rec `trimmed_rec`!

aghatpande, 2023-12-18:
Thanks much @alejoe91. Pretty straightforward the way you explain it. When I use "trimmed_rec" to apply CMR correction, will it exclude that stretch from the CMR calculation? Thanks again. Will use it as soon as i can get to it.

alejoe91, 2023-12-18:
> Thanks much @alejoe91. Pretty straightforward the way you explain it. When I use "trimmed_rec" to apply CMR correction, will it exclude that stretch from the CMR calculation? Thanks again. Will use it as soon as i can get to it.

Yes of course!

aghatpande, 2023-12-18:
Seems to work seamlessly including saving as a zarr object. @alejoe91 thank you very much. Closing the issue

closed 2023-12-18 by aghatpande