SpikeInterface/spikeinterface #2346: peak_sign in detect peaks

Antho2422, 2023-12-18:
Hi all, 

I have a question about the peak detection with the detect peak function. I was trying to plot the detection threshold on my data and visualize the detected peaks and that was my result:

[image]

When I put detect_sign on 'pos' I get only negative peaks and vice-versa. Am I doing something wrong or is there something I did not completly understand ? 

Thanks, 

Anthony

alejoe91, 2023-12-18:
Hi Anthony,

Your assumption is correct. The peaks should be positive! Can you check the distribution of `peaks["amplitude"]`?

yger, 2023-12-18:
This is strange, I'll have a look. But not that method='locally_exlusive' with a radius_um = 0 is equivalent to "by_channel" but much faster (using a numba kernel instead of numpy)

yger, 2023-12-18:
I can not reproduce the problem with main. Note that to visulize peaks, once you get peaks, you can turn them to spikes via si.NumpySorting.from_peaks(peaks, rec.get_sampling_frequency(), unit_ids=recording.channel_ids). And then, once you have a sorting object, you can use the WaveformExtractor to extract snippets and/or si.plot_spikes_on_traces to visualize the peaks on the data

Antho2422, 2023-12-18:
Note that in the histogram I plot the amplitudes of all the detected peaks (even the peaks out of the range of the 2nd plot)

Antho2422, 2023-12-18:
> I can not reproduce the problem with main. Note that to visualize peaks, once you get peaks, you can turn them to spikes via NumpySorting.from_peaks(peaks, rec.get_sampling_frequency(), unit_ids=recording.channel_ids). And then, once you have a sorting object, you can use the WaveformExtractor to extract snippets and/or plot_spikes_on_traces to visualize the peaks on the data

Okay I did not know that but I will try :) Thank you

Antho2422, 2024-01-08:
Hi @alejoe91 @yger , 

I would like to know a bit more about the peak sign detection on my data that seems to be reversed. I experienced the same issue during the sorting. To add more context I'm working with Neuralynx data. When I plot the data everything looks normal but during the peak detection I have to put the peak sign detection on positive in order to detect negative peaks values.

Another thing that I did not understand is that when I retrieve the peaks["amplitudes"] I get positive values but the values does not match the real peaks amplitude values. 

My questions are :

- Is there some known particularities about Neuralynx data that could explain this behavior ? 
- Is that an issue or should I just keep the peak detection on positive in order to detect visualy neg spikes ?
- Is that a good idea to put the peak detection on 'both' during the sorting to cover both cases ? 

I can provide a snippet of data if needed to reproduce the behavior !

Thank you, 
Anthony

yger, 2024-01-08:
A snippet of data would be helpfull, indeed, thanks. I guess it might be related to the option return_scaled (default is False) in getting peaks that is used during estimation of the noise levels, but in any case we should understand what is happening. Happy to have a look

Antho2422, 2024-01-08:
Thank you @yger, 

Here is the data snippet download link : https://instituteicm-my.sharepoint.com/:f:/g/personal/anthony_pinto_icm-institute_org/Eo-qgKE_RudBq4CYnNA5jgABUweAMKYVbkN8mvIUMKj3xQ?e=3shTjt

And here is the code you can use to reproduce what I'm doing : 
```python
import matplotlib.pyplot as plt
from spikeinterface.sortingcomponents.peak_detection import detect_peaks
from spikeinterface.preprocessing import bandpass_filter
from spikeinterface import load_extractor
import numpy as np

recording = load_extractor(r'path/to/recording')
recording = recording.frame_slice(start_frame=0, end_frame=327680)
recording = recording.channel_slice(channel_ids=['138'])
recording = bandpass_filter(recording=recording, freq_min=300, freq_max=6000, dtype='float64')
peaks = detect_peaks(
            recording=recording,
            method='by_channel',
            peak_sign='pos',
            detect_threshold=4,
            exclude_sweep_ms=0.2,
            noise_levels=None,
            random_chunk_kwargs={},
        )
traces = recording.get_traces(return_scaled=True)
time = np.linspace(0, 10, 327680)
fig = plt.figure(figsize=(30, 5))
plt.plot(time[peaks["sample_index"]], traces[peaks["sample_index"]], 'rx')
plt.plot(time, traces, linewidth=0.5, color='k')
plt.xlim(0, 10)
plt.show()
```

yger, 2024-01-08:
You sent me a folder and not directly the neuralynx data, but I guess you've tried the error is still there. I'll have a look asap

Antho2422, 2024-01-08:
Yes I sent you the base recording folder directly but if you want I can send you the data directly

Tell me if you need the raw data. Thank you

yger, 2024-01-08:
Actually, on my machine, your snippet (with the recording folder) is working as expected if you set return_scaled=False in the get_traces() call. This is because internally, detect_peaks used raw unscaled traces. I'll discuss why with @sam, but since your neuralynx data has a minus sign in the header, indeed, there is a discrepancy there...

Antho2422, 2024-01-08:
Okay, I did not know that. So for the moment I should keep the peak detection on positive ? When using SC2 for example ?

yger, 2024-01-08:
Actually, this is just a visualization issue. SC2, internally, will always work with unscaled data. So indeed either you set peak detection to pos (and compare that to your data scaled) or use negative peak detection and compare that to your data not scaled. This does not matter for the sorting, and for the quality metrics, then you can set return_scaled=True when extracting waveforms

yger, 2024-01-08:
Arg but indeed, if you have scaled data with a minus sign in this scaling factor, then you should have peak detection on pos for the sorting. You are absolutly right.... I'll discuss that with Sam

Antho2422, 2024-01-08:
Thank you @yger ! Please keep me updated on that

samuelgarcia, 2024-01-09:
Yes the peaks["amplitude"] is un scaled so you need to apply the gain and offset from recording to recover the scaled amplitude in micor volt. If your gain is negative (which is possible) then it explains the switch of the sign.

closed 2024-06-19 by zm711