SpikeInterface/spikeinterface #2347: CUDA error running kilosort3

aghatpande, 2023-12-18:
I am running Windows 11 machine with 64 GB RAM and a GPU. I'm running SpikeInterface 0.99.1 and running sorters in Docker. I got this error using both kilosort 2 as well as kilosort3 both crashing after ~ 20 min : 

> An unexpected error occurred trying to launch a kernel. The CUDA error was:
> 
> invalid configuration argument
> Error running kilosort3
a similar error has been reported in issue # 427 in the Kilosort3 repo linked here: https://github.com/MouseLand/Kilosort/issues/427

The fix they describe has actually worked for me using Kilosort3 installed independently of SpikeInterface while using a separate dataset. I'm wondering if there is a quick fix by fixing the error as described in the above issue and recreating new Dockerfiles for both kilosort2 and kilosort3? As usual, any advice / insight much appreciated.

alejoe91, 2023-12-19:
Hi @aghatpande 

Does this error occur on all recordings, or just spradically?

Unfortunately the fix is not that simple, especially if it involves changes in the KS code base that is not merged into the released version. This kind of defeats the purpose of having a versioned Docker in the first place..

In this case, I recommend using a local version of Kilosort with the proposed fix! 
You can also try to compile and build a docker image for KS yourself if you want to run via Docker. See instructions here: https://github.com/SpikeInterface/spikeinterface-dockerfiles/blob/main/kilosort-compiled/README.md

Cheers,
Alessio

aghatpande, 2023-12-20:
True, this didn't occur when I split the recording into two. Both sorters (Kilosort 2 & 3) ran successfully on the first part through Docker / SpikeInterface. They also ran successfully on other recordings again through Docker. I think the issue was specific to my recording. 
Thanks for the link and your prompt response @alejoe91. Please close this issue since it's not really an issue in most workflows using SpikeInterface.

closed 2023-12-20 by alejoe91