SpikeInterface/spikeinterface #2349: Read recording from phy folder

rory-bedford, 2023-12-18:
Hi,

I'm wondering if it's possible to read in the recording from a phy folder, after running export_to_phy, and discarding the original recording.

I can currently read a sorting object, using se.read_phy, but can't see how to read the recording. I feel like this should be possible. The phy folder contains the 'recording.dat' file, alongside channel_groups, channel_map, channel_positions, and the necessary metadata in params.py. I've started trying to write this as a function, but thought I'd check here first to see if this extractor already exists, or if there's a reason why it shouldn't exist.

Thanks!
Rory

alejoe91, 2023-12-18:
Hi Rory,

No we don't have an extractor for Phy, because essentially it's biinary data. I agree that all data are readily available, so it mike make sense to make a `read_phy_recording` function.

zm711, 2023-12-18:
@alejoe91, do we have the channel_offsets and gain? I'm looking in a phy exported folder and it looks like we are missing that info (and it's not provided any where in the files) or am I missing something?

zm711, 2024-06-19:
@alejoe91, this might be a good time to decide if we want to keep this open.

alejoe91, 2024-06-19:
Yeah you raised a good point about not having the gain and offset info. The recording can be loaded with the read_binary and if one saves a binary file with spike interface Phy will automatically point to it, and in that case the gain and offset will be available. So I would suggest to close

closed 2024-06-20 by zm711