SpikeInterface/spikeinterface #2366: Weird syntax highlighting in the `pyproject.toml`

zm711, 2023-12-21:
Does anyone know why the toml is showing this in GitHub? @alejoe91, @h-mayorquin , @samuelgarcia ? It started doing that a week ago (I think), but I finally remembered to post it here to ask everyone. I don't see a syntax error...

(edit: "remfile" is missing an optional comma I think. I tried adding it in a branch, but it didn't change the red highlighting. I've tried some searching on the web, but nothing is coming up for why they would appear red).

[image]

alejoe91, 2023-12-21:
Can it be because of the comments in `streaming_extractors`?

zm711, 2023-12-21:
We have tons of comments in the testing section and that isn't highlighted in red. So it shouldn't be that. I'm just not sure if GitHub is helping or if this is a bug in their syntax highlighting (which has happened before). I doubt "full" is a protected word, but I haven't check that yet. If people aren't complaining about install problems I guess we could just let it go and see if it was just a GitHub bug.

zm711, 2024-01-30:
@h-mayorquin,

Just wanted to rope you into this for __after the release__. I feel like you've got your finger on the pulse of things.

h-mayorquin, 2024-01-30:
I saw this @zm711 . I also think that is a bug of the with linguistic:

https://github.com/github-linguist/linguist/issues/6668

Sorry, my knowledge is lacking here. My own local one does not mark this as an error which reinforces the view that is github related and nothing wrong with our toml so I think is safe to wait until is fixed upstream.

zm711, 2024-01-30:
Thanks. Okay sounds good. I'm fine just leaving this alone and seeing what happens.

closed 2024-01-31 by zm711