SpikeInterface/spikeinterface #2372: Common reference loads all traces into memory

nadirbt, 2023-12-22:
In common_reference.py, all traces are loaded into memory which crashes my computer because my dataset is too big. Is there a workaround?        

lin 146: all_traces = self.parent_recording_segment.get_traces(start_frame, end_frame, slice(None))

alejoe91, 2023-12-22:
Hi, if you call the get traces function it is supposed to load the traces in memory, so there is no way around. Why do you need to do that?

nadirbt, 2023-12-22:
I am trying to plot the signals after applying the common reference

alejoe91, 2023-12-22:
Then you should just get a few seconds at a time ;)

nadirbt, 2023-12-22:
I give it the channel_ids parameter but the way the function is written it loads everything before indexing the channel_ids

alejoe91, 2023-12-22:
That's needed for CMR because you need to compute the median using all channels

nadirbt, 2023-12-22:
I agree, but it is not necessarily needed if one chooses a local reference or computes the average

h-mayorquin, 2024-01-02:
Average does require all the channels as well, doesn't it?

Yes, the standard workaround is to use extract shorter chunks (i.e. play with start_frame and end_frame to make the chunk shorter).

alejoe91, 2024-01-02:
Yes I agree, also reference requires all channels. 

I believe that local also needs all channels, because each channel will likely be in the neighborhood of at least one other channel. 

@nadirbt do you agree?

nadirbt, 2024-01-02:
My datasets are too big to load into memory all at once, so it would be nice to have a parameter in the common reference class to set chunk size.

For the average operation, it would be helpful to get the average chunk by chunk to not load everything all at once. This would work even if you compute the average for all channels.

In the case of a local channel, you need to subtract the referenced channel from all the channels, but you don't have to do the subtraction operation all at once. Here also, a chunk size would work.

If one of the goals of SpikeInterface is to be used for large datasets, I think this would be worth it to implement. Other functions in SpikeInterface already make use of chunked data.

alejoe91, 2024-01-02:
We usually use chunk processing all over the place. Can you share the snippet of code that you are using that comsumes too much memory? Maybe we can help you there ;)

nadirbt, 2024-01-02:
I guess I understand that I have to use chunk size when I call get_traces as @h-mayorquin pointed out. But this line prevents any slicing from happening
lin 146: all_traces = self.parent_recording_segment.get_traces(start_frame, end_frame, **slice(None)**)

alejoe91, 2024-01-02:
that line slices the recording between `start_frame` and `end_frame` already, which normally is around 1s worth of data.

Also, you normally call a `recording.get_traces`, not the `recording_segment.get_traces` that you pointed out.

h-mayorquin, 2024-01-02:
There is no argument called `chunk_size`, sorry for the misunderstanding. What I am saying is that you should chooose the parameters `start_frame` and `end_frame` to get smaller chunks. In other words, get just a piece in time of your signal that will actually fit in memory.

If your use-case is plotting you will need to do that anyway as matplotlib, for example, will require loading everything you will plot in memory.

nadirbt, 2024-01-02:
Okay, I understand. Thank you for clarifying your point. So if I understand correctly, it is not possible to plot one channel of a large dataset after it goes through the common_reference operation (at least if I use matplotlib).

alejoe91, 2024-01-02:
It is possible, but to compute that channel you still need access to all channels (except for "single" reference case, which is not so common to use)

h-mayorquin, 2024-01-02:
In your situation when only one channel fits in memory the short answer is no.

The work around would be to extract chunks (call get_traces over different values of start_frame and end_frame  until you cover your signal) and then use `np.concantenate` to get the full channel signal.

If you share with us a snippet of code that you are trying to use I would be easier for us to give you more concrete advice. Maybe you could use our widgets:

https://spikeinterface.readthedocs.io/en/latest/modules_gallery/widgets/plot_1_rec_gallery.html

To visualize one channel only as well.

More detailed answer is that we try to gaurantee that `get_traces` only uses as much memory as the one one in the chunk:of size `(end_frame - start_frame) * number_of_channels` (notice there is no channel ids). That is, we control memory allocation by time and not by channel. Reason being, signals are usually longer than "wider" so memory is usually constrain by the duration of your chunk and we focus our efforts in that direction. Within that framework your use case is sort of a corne case when a single channel fits in full in memory but not all of them.

Lastly, for my own curiosity, is there a plotting library that does not lot the data into memory before plotting?

nadirbt, 2024-01-03:
Thank you for your suggestion @h-mayorquin. I think getting different time chunks and concatenating  them seems to be the best solution here. 

I think you might use the **Bokeh** library to achieve this without loading everything in memory.

nadirbt, 2024-01-03:
Thank you @alejoe91 for your help as well.

closed 2024-01-03 by nadirbt

h-mayorquin, 2024-01-03:
I know Bokeh but I did not know it was able to do that. I will check it out. 

Glad we could help.

samuelgarcia, 2024-01-09:
Hi,
in a notebook
`plot_traces(rec, backend='ipywidgets')`
is doing lazy loading chunk by chunj to visualize a preprocessing this is usefull.