SpikeInterface/spikeinterface #2373: SilencePeriodsRecording throws an error

nadirbt, 2023-12-24:
Everytime I run a recording that was processed by the silence period function, it throws this error: 

get_random_data_chunks() got an unexpected keyword argument 'noise_levels'

I think this is because the definition of the class initializer is different from the definition of the self._kwargs argument.

yger, 2023-12-26:
Can you share a minimal example to reproduce the error? I'll have a look asap

nadirbt, 2023-12-27:
I was able to replicate the bug using this code:

import spikeinterface.full as si
from pathlib import Path

global_job_kwargs = dict(n_jobs=4, chunk_duration="1s")
si.set_global_job_kwargs(**global_job_kwargs)

local_path = Path(__file__).parent / './mearec_test_10s.h5'
recording, _ = si.read_mearec(local_path)

rec_preprocessed = si.silence_periods(recording, list_periods = [(0, 10)], mode='noise')

sorting_MS5 = si.run_sorter(sorter_name='mountainsort5', recording = rec_preprocessed, remove_existing_folder=True)

yger, 2023-12-27:
This should be fix with #2375. Indeed, noise levels were not saved as internal attributes, sorry for that

nadirbt, 2024-01-03:
Thank you for fixing this. Has the change been released yet?

yger, 2024-01-03:
No, this is only in the main branch (that you can install via git), not yet in a release

closed 2024-01-05 by nadirbt