SpikeInterface/spikeinterface #2376: Speed up non-rigid motion correction method

atombysx, 2023-12-27:
Hi, I concatenated Neuropixel 1.0 spikeGLX recordings from the same day and ran the  'non-rigid' motion correction preset.

It took 21 hours to 'detect localise' for one hour concatenated recording.

Is there any way to speed it up as I have two probes recordings over days needed sorted?

alejoe91, 2023-12-27:
Hi, are you using parallel processing? Can you share the snippet of code that you are using? One of the argument should be `n_jobs`! 

On my 32 cpu machine, correcting 2h neuropixels recording takes around 45 minutes

atombysx, 2023-12-27:
> Hi, are you using parallel processing? Can you share the snippet of code that you are using? One of the argument should be `n_jobs`!
> 
> On my 32 cpu machine, correcting 2h neuropixels recording takes around 45 minutes

Hi,

I wasn't aware of parallel processing is available for motion correct.

I was following the 'High-level API' motion correction module documentation(https://spikeinterface.readthedocs.io/en/latest/modules/motion_correction.html):
`probe0_nonrigid_accurate = si.correct_motion(recording=probe0_cat, preset="nonrigid_accurate")`

Should I just change it to this based on number of cores?
`probe0_nonrigid_accurate = si.correct_motion(recording=probe0_cat, preset="nonrigid_accurate",n_jobs=32)`

Many thanks!

alejoe91, 2023-12-27:
Yes! You can see it in the docstring here: https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.correct_motion

atombysx, 2023-12-28:
Thanks a lot! Now it's mu

closed 2023-12-28 by atombysx