SpikeInterface/spikeinterface #2378: Mountainsort 5 on Neuropixel Data

atombysx, 2023-12-30:
Hi,
I first preprocessed the neuropixel data from spikeglx with highpass filter, phase shift, concatenation, and motion correction.

I passed the data to multiple sorters including Kilosrot 3, kilosort 2.5 and moutainsort 5.

I simply ran the sorter using:
`mountainsort_test = si.run_sorter(sorter_name='mountainsort5', recording=probe0_preprocessed_corrected, output_folder=dst_folder+'/sorters/mountainsort5/probe0',docker_image=True)`

Kilosorts ran smoothly on my data but moutainsort5 has the following error that doesnt say what was wrong:
`SpikeSortingError                         Traceback (most recent call last)
Cell In[25], line 1
----> 1 mountainsort_test = si.run_sorter(sorter_name='mountainsort5', recording=probe0_preprocessed_corrected, output_folder=dst_folder+'/sorters/mountainsort5/probe0',docker_image=True)

File ~/anaconda3/envs/si_env/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:141, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    139         else:
    140             container_image = singularity_image
--> 141     return run_sorter_container(
    142         container_image=container_image,
    143         mode=mode,
    144         **common_kwargs,
    145     )
    147 return run_sorter_local(**common_kwargs)

File ~/anaconda3/envs/si_env/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py:606, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, **sorter_params)
    604 if run_error:
    605     if raise_error:
--> 606         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    607 else:
    608     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
Number of channels: 339
Number of timepoints: 9000000
Sampling frequency: 30000.0 Hz
Channel 0: [16. 40.]
Channel 1: [48. 40.]
Channel 2: [ 0. 60.]
Channel 3: [32. 60.]
Channel 4: [16. 80.]
Channel 5: [48. 80.]
Channel 6: [  0. 100.]
Channel 7: [ 32. 100.]
Channel 8: [ 16. 120.]
Channel 9: [ 48. 120.]
Channel 10: [  0. 140.]
Channel 11: [ 32. 140.]
Channel 12: [ 16. 160.]
Channel 13: [ 48. 160.]
Channel 14: [  0. 180.]
Channel 15: [ 32. 180.]
Channel 16: [ 16. 200.]
Channel 17: [ 48. 200.]
Channel 18: [  0. 220.]
Channel 19: [ 32. 220.]
Channel 20: [ 16. 240.]
Channel 21: [ 48. 240.]
Channel 22: [  0. 260.]
Channel 23: [ 32. 260.]
Channel 24: [ 16. 280.]
Channel 25: [ 48. 280.]
Channel 26: [  0. 300.]
Channel 27: [ 32. 300.]
Channel 28: [ 16. 320.]
Channel 29: [ 48. 320.]
Channel 30: [  0. 340.]
Channel 31: [ 32. 340.]
Channel 32: [ 16. 360.]
Channel 33: [ 48. 360.]
Channel 34: [  0. 380.]
Channel 35: [ 32. 380.]
[... 283 lines cut ...]
Channel 319: [  16. 3240.]
Channel 320: [  48. 3240.]
Channel 321: [   0. 3260.]
Channel 322: [  32. 3260.]
Channel 323: [  16. 3280.]
Channel 324: [  48. 3280.]
Channel 325: [   0. 3300.]
Channel 326: [  32. 3300.]
Channel 327: [  16. 3320.]
Channel 328: [  48. 3320.]
Channel 329: [   0. 3340.]
Channel 330: [  32. 3340.]
Channel 331: [  16. 3360.]
Channel 332: [  48. 3360.]
Channel 333: [   0. 3380.]
Channel 334: [  32. 3380.]
Channel 335: [  16. 3400.]
Channel 336: [  48. 3400.]
Channel 337: [   0. 3420.]
Channel 338: [  32. 3420.]
Loading traces
Detecting spikes

Adjacency for detect spikes with channel radius 200
[[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 21], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 23], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 25], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 27], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 29], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 31], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 33], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 35], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 37], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38], [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 39], [0, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40], [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 41], [2, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42], [3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 43], [4, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44], [5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 45], [6, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46], [7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 47], [8, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48], [9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 49], [10, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50], [11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 51], [12, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52], [13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 53], [14, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54], [15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 55], [16, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39 [... cut ...] , 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 157], [118, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158], [119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141,

magland, 2024-01-02:
It's hard to say what could be going wrong. Maybe it's running out of memory?

It might be helpful to try with a smaller subset of the data -- perhaps and/or a channel slice or a time slice.

atombysx, 2024-01-02:
> @atombysx
> 
> You can do that with this (e.g., running on 10 minutes data):
> 
> ```
> probe0_preprocessed_corrected_stub = probe0_preprocessed_corrected.frame_slice(
>     start_frame=0, end_frame=int(600 * probe0_preprocessed_corrected.sampling_frequency)
> )
> ```
> 
> and then use the `run_sorter` function with the `probe0_preprocessed_corrected_stub` object

Hi I just tried what you said:
`probe0_preprocessed_corrected_stub = probe0_preprocessed_corrected.frame_slice(
    start_frame=0, end_frame=int(600 * probe0_preprocessed_corrected.sampling_frequency)
)`
`job_list = [  {'sorter_name':'mountainsort5','recording':probe0_preprocessed_corrected_stub,'output_folder':dst_folder+'/probe0/sorters/mountainsort5/','docker_image':True}]
probe0_fragment_moutainsort5 = si.run_sorter_jobs(job_list = job_list, engine = 'loop')`

The same error happens.

It was only successful when I reduced it to 1 min slice....

We have 64GB RAM on this machine,

JoeZiminski, 2024-06-20:
@magland did you have any further thoughts on this? @atombysx were you able to find any other workaround? 🤔

magland, 2024-06-20:
> @magland did you have any further thoughts on this? @atombysx were you able to find any other workaround? 🤔

I'm not sure. I guess we still don't know if it's a memory issue...

zm711, 2024-12-05:
Since this is ultimately an issue interacting with MS5 I'll close this here. Feel free to open issues on the sorter repo directly for more help.

closed 2024-12-05 by zm711