SpikeInterface/spikeinterface #2383: Difference between removing duplicates methods

nadirbt, 2024-01-03:
Hello,
I am trying to understand the difference between different remove duplicates methods: remove_duplicates_via_matching, remove_duplicates_via_dip, remove_duplicates. I am trying to see which one to use for my datatsets.

yger, 2024-01-03:
The sortingcomponents toolbox needs some cleaning. These are various attempts to remove "duplicates", i.e. the fact that after the clustering, some templates might be found twice, with or without temporal lags, and also that we might have mixtures of templates. The best method so far (for SC2) is to remove via matching:  we will ensure that no templates in the final dict can be explain as a linear sum of others (using the same orthogonal matching engine as in the fitting part). Any linear sum is removed. the Dip method is what was used originially in SC, and is also used in MS5. You look at all the clusters (templates are only the centroids) and you ask via a dip test if, for a given pair of templates, the distribution looks unimodal (and thus we should merge) or bimodal. Remove_duplicates is simply I guess a method on the cosine similarity: if two templates are similar (wrt cosine metric) they are removed.

nadirbt, 2024-01-03:
Thank you for the explanation :)

closed 2024-01-03 by nadirbt