SpikeInterface/spikeinterface #2387: get_chunk_with_margin() should accept return_scaled

yger, 2024-01-08:
As mentionned in #2346, there is slight problem with the sortingcomponents. Currently, because of the internal function get_chunk_with_margin that do not accept return_scaled, all internal operations are performed under the hood on raw data (no scaling). This is quite often not a major issue, but if some fileformat have a negative gain, then the peaks are all upside down internally for the components. We could simply extend this function (and therefore all the node_pipelines) to accept a return_scaled argument, and leave it as default to True, as much as possible?

alejoe91, 2024-01-08:
@yger this is very tricky. The reason why `return_scaled` is only handled by the `Recording` is that if you have a processing chain you only want to apply scaling once! 

In my opinion, for the `Neuralynx` issue, the user should set the sign to positive, or sorters could do that automatically if they detect a negative gain. What do you think?

yger, 2024-01-08:
Hum... I did not consider such a problem. I'll keep thinking about it, but if no better solution, then what you are suggesting is indeed the way to proceed. If a negative gain is detected, then we should flip peak signs internally.

yger, 2024-01-08:
Actually, new proposal before discussion. Since the problem is only in the run_node_pipeline, mostly used in components, then maybe in the node pipeline, when we want to access the (final, i.e. the end of preprocessing chain) traces for the nodes, we could use a different function (get_traces_with_margin) that will make an internal call to get_traces (and thus get the scaling right). So we keep get_chunk_with_margin as a low-level function, not making any rescaling, and we have a new one (only used in the node pipeline) get scaled data. To be discussed