SpikeInterface/spikeinterface #2390: Compute_template_metrics() TypeError: compute_template_metrics() got an unexpected keyword argument 'include_multi_channel_metrics'

taningh86, 2024-01-08:
I used the following code : 
```
spost.compute_template_metrics(we, load_if_exists= True, metric_names=['peak_to_valley', 'peak_trough_ratio', 'half_width',  
repolarization_slope', 'recovery_slope'], peak_sign= 'neg', upsampling_factor= 10, sparsity= sparsity, include_multi_channel_metrics= False, metrics_kwargs=None)
```
And, it gives the TypeError: compute_template_metrics() got an unexpected keyword argument 'include_multi_channel_metrics'. I get the same error for 'metrics_kwargs' after I omit  'include_multi_channel_metrics' altogether. I know I can pass additional arguments to 'metrics_kwargs' and did try "metrics_kwargs={'recovery_window_ms': 0.7}" but I get the similar type error. 
I am assuming my code is correct here but not sure where is the error originating from. Will appreciate any suggestion or resolution for this error.

Thanks
Jimmy

alejoe91, 2024-01-08:
Jimmy can you make sure you have version 0.99.1?

taningh86, 2024-01-08:
I have 0.98.2. I will update now and check if it works. Will update shortly. Thanks Alessio!

taningh86, 2024-01-09:
It works with the 0.99.1 version.
On a side note, at first I updated to 0.100.0.dev1 version and it was really slow.

closed 2024-01-09 by alejoe91