SpikeInterface/spikeinterface #2391: spikes exceeding recording duration?

a-zmz, 2024-01-09:
Hi,

I have been sorting a new batch of recordings, and got 
```ValueError: The sorting object has spikes exceeding the recording duration. You have to remove those spikes with the `spikeinterface.curation.remove_excess_spikes()` function``` in some of them, could you please advise why this happened...

I installed spikeinterface 0.99.1 on pip, and I am running fedora 37.

Thanks in advance!

alejoe91, 2024-01-09:
Sometimes Kilosort (I guess) produces spikes beyonf the length of the recording. I think this is because the `temp_wh` file that it produces is padded. 

You just need to run this to remove these excess spikes:
```
from spikeinterface.curation import remove_excess_spikes

sorting_clean = remove_excess_spikes(sorting=sorting_bad, recording=recording_used_for_sorting)
```

a-zmz, 2024-01-09:
hi, thanks for advising. I was just worried that has anything to do with my data acquisition, or large amount of good spikes will be removed. Good to know it's the kilosort (yea I use ks3). I have checked one of the recordings that threw the error, there were just one spike needed to be removed.

Thanks again!

closed 2024-01-09 by a-zmz