SpikeInterface/spikeinterface #2392: Troubles reading brw file

DianaNigrisoli, 2024-01-10:
Hi, I'm having issues reading a brw file. I have installed spikeinterface[full] package in my virtualenv. This is my code: 
` import spikeinterface.extractors as se
full_raw_rec = se.read_biocam('/Users/diana/Desktop/Organoid_Baseline_Spiking_activity.brw')
`

The traceback error is the following: 
`---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[15], [line 6](vscode-notebook-cell:?execution_count=15&line=6)
      [1](vscode-notebook-cell:?execution_count=15&line=1) import h5py
      [3](vscode-notebook-cell:?execution_count=15&line=3) # with h5py.File('C:\\Users\\diana\\Desktop\\Organoid_Baseline_Spiking_activity.brw','r+') as ff:
      [4](vscode-notebook-cell:?execution_count=15&line=4) #    dset = ff.create_group("MyGroup")
----> [6](vscode-notebook-cell:?execution_count=15&line=6) full_raw_rec = se.read_biocam('/Users/diana/Desktop/Organoid_Baseline_Spiking_activity.brw')

File [~\Desktop\spikeinterface\src\spikeinterface\extractors\neoextractors\biocam.py:47](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:47), in BiocamRecordingExtractor.__init__(self, file_path, mea_pitch, electrode_width, stream_id, stream_name, block_index, all_annotations)
     [36](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:36) def __init__(
     [37](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:37)     self,
     [38](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:38)     file_path,
   (...)
     [44](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:44)     all_annotations=False,
     [45](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:45) ):
     [46](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:46)     neo_kwargs = self.map_to_neo_kwargs(file_path)
---> [47](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:47)     NeoBaseRecordingExtractor.__init__(
     [48](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:48)         self, stream_id=stream_id, stream_name=stream_name, all_annotations=all_annotations, **neo_kwargs
     [49](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:49)     )
     [51](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:51)     # load probe from probeinterface
     [52](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:52)     probe_kwargs = {}

File [~\Desktop\spikeinterface\src\spikeinterface\extractors\neoextractors\neobaseextractor.py:185](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:185), in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    [156](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:156) def __init__(
    [157](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:157)     self,
    [158](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:158)     stream_id: Optional[str] = None,
   (...)
    [163](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:163)     **neo_kwargs: Dict[str, Any],
    [164](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:164) ) -> None:
    [165](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:165)     """
    [166](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:166)     Initialize a NeoBaseRecordingExtractor instance.
    [167](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:167) 
   (...)
    [182](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:182) 
    [183](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:183)     """
--> [185](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:185)     _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
    [187](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:187)     kwargs = dict(all_annotations=all_annotations)
    [188](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:188)     if block_index is not None:

File [~\Desktop\spikeinterface\src\spikeinterface\extractors\neoextractors\neobaseextractor.py:25](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:25), in _NeoBaseExtractor.__init__(self, block_index, **neo_kwargs)
     [23](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:23) def __init__(self, block_index, **neo_kwargs):
     [24](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:24)     if not hasattr(self, "neo_reader"):  # Avoid double initialization
---> [25](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:25)         self.neo_reader = self.get_neo_io_reader(self.NeoRawIOClass, **neo_kwargs)
     [27](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:27)     if self.neo_reader.block_count() > 1 and block_index is None:
     [28](https://file+.vscode-resource.vscode-cdn.net/c%3A/Users/diana/Desktop/spikeinterface/~/Desktop/spikeinterface/src/spikeinterface/extractors/neoextractors/neobaseextractor.py:28)         raise Exception(
[... cut ...]
---> [64](file:///C:/Users/diana/Desktop/spikeinterface/.venv/Lib/site-packages/h5py/_hl/group.py:64) gid = h5g.create(self.id, name, lcpl=lcpl, gcpl=gcpl)
     [65](file:///C:/Users/diana/Desktop/spikeinterface/.venv/Lib/site-packages/h5py/_hl/group.py:65) return Group(gid)

File h5py\_objects.pyx:54, in h5py._objects.with_phil.wrapper()

File h5py\_objects.pyx:55, in h5py._objects.with_phil.wrapper()

File h5py\h5g.pyx:166, in h5py.h5g.create()

ValueError: Unable to synchronously create group (no write intent on file)`

alejoe91, 2024-01-10:
Hi Diana,

We have support for the BRW v4.x, but it's not released yet. (see https://github.com/NeuralEnsemble/python-neo/pull/1326)

You can fix it by installing neo from source:
```
pip install git+https://github.com/NeuralEnsemble/python-neo.git
```

nicosese, 2024-01-10:
yup

alejoe91, 2024-01-10:
We have an open PR in probeinterface for the new BRW version.

Can you try to install `probeinterface` from this PR? 
https://github.com/SpikeInterface/probeinterface/pull/216

nicosese, 2024-01-10:
Apparently missing channel indices. I get this:
`AssertionError                            Traceback (most recent call last)
Cell In[4], [line 1](vscode-notebook-cell:?execution_count=4&line=1)
----> [1](vscode-notebook-cell:?execution_count=4&line=1) rawrec = se.read_biocam('[/Users/nike/Desktop/15_Accura_HumanOrganoid_ISOM2XAV_DICV116_stim](https://file+.vscode-resource.vscode-cdn.net/Users/nike/Desktop/15_Accura_HumanOrganoid_ISOM2XAV_DICV116_stim)+4AP.brw')

File [~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:58](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:58), in BiocamRecordingExtractor.__init__(self, file_path, mea_pitch, electrode_width, stream_id, stream_name, block_index, all_annotations)
     [56](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:56)     probe_kwargs["electrode_width"] = electrode_width
     [57](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:57) probe = probeinterface.read_3brain(file_path, **probe_kwargs)
---> [58](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:58) self.set_probe(probe, in_place=True)
     [59](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:59) self.set_property("row", self.get_property("contact_vector")["row"])
     [60](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/extractors/neoextractors/biocam.py:60) self.set_property("col", self.get_property("contact_vector")["col"])

File [~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:96](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:96), in BaseRecordingSnippets.set_probe(self, probe, group_mode, in_place)
     [94](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:94) probegroup = ProbeGroup()
     [95](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:95) probegroup.add_probe(probe)
---> [96](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:96) return self.set_probes(probegroup, group_mode=group_mode, in_place=in_place)

File [~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:144](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:144), in BaseRecordingSnippets.set_probes(self, probe_or_probegroup, group_mode, in_place)
    [141](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:141)     raise ValueError("must give Probe or ProbeGroup or list of Probe")
    [143](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:143) # handle not connected channels
--> [144](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:144) assert all(
    [145](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:145)     probe.device_channel_indices is not None for probe in probegroup.probes
    [146](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:146) ), "Probe must have device_channel_indices"
    [148](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:148) # this is a vector with complex fileds (dataframe like) that handle all contact attr
    [149](https://file+.vscode-resource.vscode-cdn.net/Users/nike/spikevenv/~/spikevenv/spikeinterface/src/spikeinterface/core/baserecordingsnippets.py:149) probe_as_numpy_array = probegroup.to_numpy(complete=True)

AssertionError: Probe must have device_channel_indices`

alejoe91, 2024-01-10:
Let me fix that real quick ;)

alejoe91, 2024-01-10:
Actually, it should have device indices. 

Can you try this?
```
import probeinterface as pi

probe = pi.read_3brain("....brw")
print(probe.device_channel_indices)
```

nicosese, 2024-01-10:
Says none :(

alejoe91, 2024-01-10:
Can you share the file with me? It seems weird because the last line of the `read_3brain` should set the `device_channel_indices`. (https://github.com/SpikeInterface/probeinterface/blob/368d3f65fa7c2d92b4db83664b0988496fd4921e/src/probeinterface/io.py#L596)

What is the output of `probe.contact_positions.shape`?

nicosese, 2024-01-10:
My bad, I was playing around with the code and commented the lines that assigned the channel indices. Now it works :)

closed 2024-01-31 by zm711