SpikeInterface/spikeinterface #2393: Extracting Wavefoems

wardfadel, 2024-01-10:
Hi Alessio  and wish you with the team a happy new year!
I am trying to extract the waveforms from a neuropixels 384 channels probe. 
The extracted waveforms are miscomputed, and the sparsity waveforms information is wrong compared to the sorted Phy ones.
May I ask for help, and here is the code:

```ruby

X_probe= pi.generate_multi_columns_probe(
    num_columns= 4,
    num_contact_per_column= 96,
    xpitch= [12+4],
    ypitch= 40,
    y_shift_per_column= np.array([20, 0,20,0]),
    contact_shapes= "square",
    contact_shape_params= {"width": 12})

channel_indices = np.arange(384)

channel_indices = np.arange(384).reshape((4, -1), order='F')

desired_order_within_column = [0, 2, 1, 3]  

for i in range(channel_indices.shape[1]):
    channel_indices[:, i] = channel_indices[desired_order_within_column, i]

channel_indices = np.flipud(channel_indices)

channel_indices = channel_indices.flatten()

X_probe.set_device_channel_indices(channel_indices)
X_probe.set_contact_ids(channel_indices)
plot_probe(X_probe, with_contact_id=True,  with_device_index = True)
plt.show()

recording = si.read_binary(recording_file, num_channels=384, sampling_frequency=30000, dtype="int16",gain_to_uV = 2.34, offset_to_uV = 0, is_filtered=False)
recording = spre.bandpass_filter(recording, freq_min=300, freq_max=6000)
recording = recording.set_probe(Ricsi_probe , group_mode="by_probe", in_place=False) #in_place= True

sorting = si.read_phy(base_folder, exclude_cluster_groups = ["noise", "mua"])
 we = si.extract_waveforms(recording, sorting, allow_unfiltered=False, folder=waveform_folder, mode="folder",   return_scaled=True)
 
we.set_recording(recording=recording, allow_unfiltered=False)

# unit_locations = si.compute_unit_locations(we)
# spike_locations = si.compute_spike_locations(we)
# spike_amplitudes = si.compute_spike_amplitudes(we)

# correlograms = si.compute_correlograms(we)
# principal_components = si.compute_principal_components(we)
# quality_metrics = si.compute_quality_metrics(we)
# template_metrics = si.compute_template_metrics(we)

```
[image]
[image]
[image]
[image]

wardfadel, 2024-01-11:
```ruby

Hi Alessio, it worked this way, but why it created problems the other way when adding a similar probe information,i.e, X_probe.
secondly I relaized there are two probe electrode positions apart from X_probe, one is: 

channel_position = np.load(channel_position_file)
print (channel_position)
and this is the first few channel locations:

and the other is from  print (recording.get_channel_locations())
[[  43.   20.]
 [  11.   20.]
 [  59.   40.]
 [  27.   40.]
 [  43.   60.]
 [  11.   60.]
 [  59.   80.]
 [  27.   80.]
 [  43.  100.]
 [  11.  100.]
 [  59.  120.]
 [  27.  120.]
 and the other is 

[[  16.    0.]
 [  48.    0.]
 [   0.   20.]
 [  32.   20.]
 [  16.   40.]
 [  48.   40.]
 [   0.   60.]
 [  32.   60.]
 [  16.   80.]
 [  48.   80.]
 [   0.  100.]
 [  32.  100.]
 [  16.  120.]
 [  48.  120.]
 why there is such a shift althogh it looks that it hasn`t affected the end results.

Finally I attach the  w = sw.plot_unit_probe_map(we, unit_ids = [3])  and it is showing periodically colored channels rather than a one which shows high value (yellow) only for  unit_ids = [3.](![unit_probe_map](https://github.com/SpikeInterface/spikeinterface/assets/45574964/2b2c42a0-cc27-415e-858c-66c66cd070ac))

```

zm711, 2024-04-11:
@alejoe91 (just going through old issues), this looks off to me. Then channels seem to just be repeating, but not super familiar with this probe if you want to take a look at some point.

alejoe91, 2024-04-11:
Yeah I think this must be due to something wrong with the probe generation and it seems to be solved when using `read_spikeglx`. So I would close it.

@wardfadel feel free to reopen or open another issue if needed

closed 2024-04-11 by zm711