SpikeInterface/spikeinterface #2394: Smoothing using filter_gaussian

TomBugnon, 2024-01-10:
Hi @DradeAW @alejoe91 

I would like to be able to smooth or low-pass filter a recording. Bandpass recording with near-0 freq_min doesn't work in my case because I don't want to normalize across distant epochs.

I understand low-pass filtering is not supported because it is incompatible with chunking, and there is unfortunately no SmoothRecording (or the like) allowing to apply a rolling mean or median.

However if I understand properly, the GaussianFilterRecording applies convolution with a difference of gaussians to act as a bandpass filter. Currently, `freq_min` is enforced to be strictly positive, but it seems like it could be easily modified to support simple gaussian smoothing with `freq_min=None`. Do you think that would work @DradeAW ?
Best

DradeAW, 2024-01-10:
Yes, it would require some changes to the code but we could make `freq_min` and `freq_max` optional!

Then on this line (https://github.com/SpikeInterface/spikeinterface/blob/03126a3abbd95621bac0d6f76190326d9081dba1/src/spikeinterface/preprocessing/filter_gaussian.py#L80), we simply have to put only one of them if the other one is `None` :)

TomBugnon, 2024-01-10:
great thanks, I'll give it a shot!

Also from what I'm seeing so far, get_traces seems much much slower than for the regular bandpassfilterrecording... is that really what it takes?

DradeAW, 2024-01-10:
Yes Gaussian filtering is inherently slower than butterworth or bessel filtering.

My implementation is actually a lot faster than the scipy implementation from the testing I did.

zm711, 2024-02-06:
Since the GaussianFilter was merged I'll close this as completed by #2397.