SpikeInterface/spikeinterface #2396: OSError: too many open files

a-zmz, 2024-01-10:
Hi,

I tried to extract waveform from one sorted session, and it keeps crashing:

```
> 20231201_az_WDAN07 imec0.ap waveforms not extracted, extracting now.                                                                                                                                                               [210/738]
                                                                                                                                                                                                                                              
extract waveforms shared_memory multi buffer: 100%|██████████████████████████████| 3645/3645 [00:00<00:00, 8958.39it/s]                                                                                                                       
extract waveforms shared_memory multi buffer: 100%|█████████████████████████████| 3645/3645 [00:00<00:00, 10124.65it/s]                                                                                                                       
extract waveforms shared_memory multi buffer: 100%|█████████████████████████████| 3645/3645 [00:00<00:00, 10273.34it/s]                                                                                                                       
extract waveforms shared_memory multi buffer: 100%|██████████████████████████████| 3645/3645 [00:00<00:00, 8616.66it/s]                                                                                                                       
extract waveforms shared_memory multi buffer: 100%|█████████████████████████████| 3645/3645 [00:00<00:00, 10356.94it/s]                                                                                                                       
extract waveforms shared_memory multi buffer: 100%|█████████████████████████████| 3645/3645 [00:00<00:00, 10295.60it/s]                                                                                                                       
---------------------------------------------------------------------------                                                                                                                                                                   
AssertionError                            Traceback (most recent call last)                                                                                                                                                                   
File /opt/neuropixels/pixels/pixels/behaviours/base.py:916, in Behaviour.sort_spikes(self, CatGT_app, old)                                                                                                                                    
    915 try:                                                                                                                                                                                                                                  
--> 916     waveforms = si.WaveformExtractor.load_from_folder(                                                                                                                                                                                
    917         folder=cache,                                                                                                                                                                                                                 
    918         sorting=ks3_output,                                                                                                                                                                                                           
    919     )                                                                                                                                                                                                                                 
    920     print(f"> {self.name} {stream_id} waveforms extracted, now it is loaded.\n")                                                                                                                                                      
                                                                                                                                                                                                                                              
File ~/.conda/envs/pixels/lib/python3.10/site-packages/spikeinterface/core/waveform_extractor.py:141, in WaveformExtractor.load_from_folder(cls, folder, with_recording, sorting)                                                             
    140 folder = Path(folder)                                                                                                                                                                                                                 
--> 141 assert folder.is_dir(), f"This waveform folder does not exists {folder}"                                                                                                                                                              
    143 if not with_recording:                                                                                                                                                                                                                
    144     # load                                                                                                                                                                                                                            
                                                                                                                                                                                                                                              
AssertionError: This waveform folder does not exists /home/amz/interim/npx/20231201_az_WDAN07/cache_0                                                                                                                                         
                                                                                                                                                                                                                                              
During handling of the above exception, another exception occurred:                                                                                                                                                                           
                                                                                                                                                                                                                                              
OSError                                   Traceback (most recent call last)                                                                                                                                                                   
File ~/Documents/git/analysis/analysis/preprocessing/preprocessing.py:31, in <module>
     17 """                                                
     18 # step 1: do CatGT                                 
     19 az_utils.run_catgt(                                
   (...)                                                   
     27                                                    
     28 """                                                
     29 # step 2: spikesorting
     30 #exp.sort_spikes()                                 
---> 31 exp.sort_spikes(CatGT_app=catgt_dir)
     32 stop = timeit.default_timer()
     33 print(f"> Spike sorting of {exp} took {(stop - start)/60} mins.")

File /opt/neuropixels/pixels/pixels/experiment.py:126, in Experiment.sort_spikes(self, CatGT_app)
    123 for i, session in enumerate(self.sessions)
    124     print(">>>>> Sorting spikes for session {} ({} / {})"
    125            .format(session.name, i + 1, len(self.sessions)))
--> 126     session.sort_spikes(CatGT_app=CatGT_app)

File /opt/neuropixels/pixels/pixels/behaviours/base.py:925, in Behaviour.sort_spikes(self, CatGT_app, old)
    922     print(f"> {self.name} {stream_id} waveforms not extracted, extracting now.\n")
    923     #if ks3_output.count_total_num_spikes()
    924     # extract waveforms
--> 925     waveforms = si.extract_waveforms(
    926         recording=concat_rec, #recording=test, # for testing
    927         sorting=ks3_output,
[... cut ...]
  File "/home/amz/.conda/envs/pixels/lib/python3.10/asyncio/selector_events.py", line 87, in close
    self._close_self_pipe()
  File "/home/amz/.conda/envs/pixels/lib/python3.10/asyncio/selector_events.py", line 94, in _close_self_pipe
    self._remove_reader(self._ssock.fileno())
AttributeError: '_UnixSelectorEventLoop' object has no attribute '_ssock'                                                                                                                     
```

Could you please advise?

Thanks very much!

a-zmz, 2024-01-10:
tried to run with python rather than ipython, same error occurred...

samuelgarcia, 2024-01-12:
Hi @a-zmz and Zach.
Yes at the moment `ulimit -n 8192` is the way to go on linux.

And yes, the futur SortingResult will have a new etension for extracting wevform that will be mono buffer and so will avoid this error.

zm711, 2024-04-30:
Since the waveformextractor is now old api, moving forward the best solution for this issue is to upgrade to the `SortingAnalzyer` which has the monobuffer.

closed 2024-04-30 by zm711