SpikeInterface/spikeinterface #2401: Issues Extracting Waveforms for Subset of Units

litzj01, 2024-01-12:
I am getting an error when trying to extract the waveforms only for a subset of units. I have recordings with huge numbers of units and spikes, that make it impossible to extract the waveforms for all units without having extremely large file sizes (even with sparsity). I saw someone had raised a similar issue (#330), and the suggested solution was to create a sub-sorter object first which is what I am trying to do, but it is not working.

Here is the code I am running:
```
#Get recording object 
recording = se.read_biocam('D:/Temporary_Storage_for_Cdrive/060623_Primate/06.06.23_Primate_Reye_OD_ST_Stimuli_01.brw') 

#Get sorting object 
sorting = ss.read_sorter_folder(output_folder='D:/Temporary_Storage_for_Cdrive/060623_Primate/herdingspikes_output', register_recording=True, sorting_info=True) 
 
#Load unit ids 
unit_ids=sorting.unit_ids 

#Load channel ids 
channel_ids=recording.channel_ids 

#Create array of subset of units
OnCells = [1,4,6,7,10,13,19,20,24,27,30,36,38,44,46,47,51,54,70,71,75,80,94,104,107,110,118,137,139,153,175,182,184,189,213,217,218,224,226,233,238,248,284,298,301,307,327,331,378,404,487]
OffCells = [0,42,56,59,66,89,108,114,116,133,149,155,156,158,160,174,176,178,190,192,193,203,209,228,253,268,283,296,316,338,343,361,369,403,412,414,423,449,451,484,523,569,586,823]
OnOffCells =[2,3,5,8,11,17,18,22,23,25,26,29,35,37,39,40,48,50,52,55,62,67,69,74,76,77,79,83,85,98,99,105,111,112,115,117,132,135,140,141,142,143,145,150,164,167,168,169,171,179,183,185,186,188,191,195,198,200,205,208,216,220,223,237,247,251,254,255,256,258,278,280,282,302,333,337,349,377,391,405,455,497,639,4297,6582,7343]

#Create new sorting objects with unit subsets
sortingON=sorting.select_units(unit_ids=OnCells)
sortingON.save(folder="D:/Temporary_Storage_for_Cdrive/060623_Primate/" + 'OnCells')

sortingOFF=sorting.select_units(unit_ids=OffCells)
sortingOFF.save(folder="D:/Temporary_Storage_for_Cdrive/060623_Primate/" + 'OffCells')

sortingONOFF=sorting.select_units(unit_ids=OnOffCells)
sortingONOFF.save(folder="D:/Temporary_Storage_for_Cdrive/060623_Primate/" + 'OnOffCells', )

#Extract the waveforms for the On cells
we_sparseON = si.extract_waveforms(recording, sortingON, "D:/Temporary_Storage_for_Cdrive/060623_Primate/OnCells/" + 'waveforms', overwrite=True, allow_unfiltered=True, sparse=True, method="radius", radius_um=70, **global_job_kwargs) 
print("Extracting waveforms:", we_sparseON)
# %%
amplitudes = spost.compute_spike_amplitudes(we_sparseON) 
unit_locations = spost.compute_unit_locations(we_sparseON) 
spike_locations = spost.compute_spike_locations(we_sparseON) 
correlograms, bins = spost.compute_correlograms(we_sparseON) 
similarity = spost.compute_template_similarity(we_sparseON)
```

Then I get the following error:
```

BrokenProcessPool                         Traceback (most recent call last)
Cell In[13], [line 1](vscode-notebook-cell:?execution_count=13&line=1)
----> [1](vscode-notebook-cell:?execution_count=13&line=1) we_sparseON = si.extract_waveforms(recording, sortingON, "[D:/Temporary_Storage_for_Cdrive/060623_Primate/OnCells/](file:///D:/Temporary_Storage_for_Cdrive/060623_Primate/OnCells/)" + 'waveforms', overwrite=True, allow_unfiltered=True, sparse=True, method="radius", radius_um=70, **global_job_kwargs, unit_ids=OnCells) 
      [2](vscode-notebook-cell:?execution_count=13&line=2) print("Extracting waveforms:", we_sparseON)
      [3](vscode-notebook-cell:?execution_count=13&line=3) # %%

File [c:\Users\litzj\miniconda\envs\spikeinterface\lib\site-packages\spikeinterface\core\waveform_extractor.py:1593](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1593), in extract_waveforms(recording, sorting, folder, mode, precompute_template, ms_before, ms_after, max_spikes_per_unit, overwrite, return_scaled, dtype, sparse, sparsity, num_spikes_for_sparsity, allow_unfiltered, use_relative_path, seed, load_if_exists, **kwargs)
   [1591](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1591)         assert all(ch in recording.channel_ids for ch in channels), "Invalid channel ids in sparsity"
   [1592](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1592) elif sparse:
-> [1593](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1593)     sparsity = precompute_sparsity(
   [1594](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1594)         recording,
   [1595](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1595)         sorting,
   [1596](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1596)         ms_before=ms_before,
   [1597](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1597)         ms_after=ms_after,
   [1598](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1598)         num_spikes_for_sparsity=num_spikes_for_sparsity,
   [1599](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1599)         allow_unfiltered=allow_unfiltered,
   [1600](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1600)         **estimate_kwargs,
   [1601](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1601)         **job_kwargs,
   [1602](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1602)     )
   [1603](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1603) else:
   [1604](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1604)     sparsity = None

File [c:\Users\litzj\miniconda\envs\spikeinterface\lib\site-packages\spikeinterface\core\waveform_extractor.py:1718](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1718), in precompute_sparsity(recording, sorting, num_spikes_for_sparsity, unit_batch_size, ms_before, ms_after, allow_unfiltered, **kwargs)
   [1716](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1716) local_ids = unit_ids[sl]
   [1717](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/site-packages/spikeinterface/core/waveform_extractor.py:1717) local_sorting = sorting.select_units(local_ids)
[... 99 lines cut ...]
File [c:\Users\litzj\miniconda\envs\spikeinterface\lib\concurrent\futures\_base.py:610](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:610), in Executor.map(self, fn, timeout, chunksize, *iterables)
    [607](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:607) if timeout is not None:
    [608](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:608)     end_time = timeout + time.monotonic()
--> [610](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:610) fs = [self.submit(fn, *args) for args in zip(*iterables)]
    [612](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:612) # Yield must be hidden in closure so that the futures are submitted
    [613](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:613) # before the first iterator value is required.
    [614](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:614) def result_iterator():

File [c:\Users\litzj\miniconda\envs\spikeinterface\lib\concurrent\futures\_base.py:610](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:610), in <listcomp>(.0)
    [607](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:607) if timeout is not None:
    [608](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/_base.py:608)     end_time = timeout + time.monotonic()
[... cut ...]
--> [720](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/process.py:720)         raise BrokenProcessPool(self._broken)
    [721](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/process.py:721)     if self._shutdown_thread:
    [722](file:///C:/Users/litzj/miniconda/envs/spikeinterface/lib/concurrent/futures/process.py:722)         raise RuntimeError('cannot schedule new futures after shutdown')

BrokenProcessPool: A child process terminated abruptly, the process pool is not usable anymore
```

zm711, 2024-01-12:
@litzj01

What are the `job_kwargs` you are using? What is the version of spikeinterface? What is the version number of threadpoolctl? (ie you can just do a conda list in your spikeinterface environment and post that all here).

And based on your initial statement you were able to get this to work on your full data, but it was too big (the result files took up too much space I mean) or even if you try your full dataset it fails?

litzj01, 2024-01-12:
Hello, thanks for your reply! The spike interface version is 0.99.0.dev0. The job_kwargs are just
```
 global_job_kwargs = dict(n_jobs=12) 
si.set_global_job_kwargs(**global_job_kwargs)
```
I have successfully extracted the waveforms for all units on other slightly smaller datasets. However, this raw data file is 92 GB and there are 12,374 detected units with over 14 million spikes total. I did try extracting the waveforms for all units in this dataset, but it failed because the code was trying to use too many threads which exceeded the limit my ssh server allows, so it likely wasn't related to this issue. Now because of that limitation, I am trying to extract the subset of waveforms on my local computer after running the sort in a ssh client and downloading the sorter output. 

Here are all the packages in my environment:

```
# Name                    Version                   Build  Channel
aiohttp                   3.8.6                    pypi_0    pypi
aiosignal                 1.3.1                    pypi_0    pypi
altair                    5.1.2                    pypi_0    pypi
asciitree                 0.3.3                    pypi_0    pypi
asttokens                 2.4.0              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.3                    pypi_0    pypi
attrs                     23.1.0                   pypi_0    pypi
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
blas                      1.0                         mkl
bottleneck                1.3.5           py310h9128911_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
bzip2                     1.0.8                he774522_0
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cbor2                     5.5.1                    pypi_0    pypi
certifi                   2023.7.22                pypi_0    pypi
cffi                      1.16.0                   pypi_0    pypi
charset-normalizer        3.3.0                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
cloudpickle               3.0.0                    pypi_0    pypi
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
comm                      0.1.4              pyhd8ed1ab_0    conda-forge
contourpy                 1.0.5           py310h59b6b97_0
cryptography              41.0.5                   pypi_0    pypi
cuda-python               12.2.0                   pypi_0    pypi
cycler                    0.11.0             pyhd3eb1b0_0
cython                    3.0.0           py310h2bbff1b_0
dask                      2023.10.1                pypi_0    pypi
debugpy                   1.6.7           py310hd77b12b_0
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
distinctipy               1.2.3                    pypi_0    pypi
distributed               2023.10.1                pypi_0    pypi
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
fasteners                 0.19                     pypi_0    pypi
figurl                    0.2.18                   pypi_0    pypi
figurl-jupyter            0.2.8                    pypi_0    pypi
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.12.1               ha860e81_0
frozenlist                1.4.0                    pypi_0    pypi
fsspec                    2023.10.0                pypi_0    pypi
giflib                    5.2.1                h8cc25b3_3
glib                      2.69.1               h5dc1a3c_2
h5py                      3.9.0           py310hfc34f40_0
hdf5                      1.12.1               h51c971a_3
hdmf                      3.11.0                   pypi_0    pypi
herdingspikes             0.3.102                  pypi_0    pypi
icc_rt                    2022.1.0             h6049295_2
icu                       58.2                 ha925a31_3
idna                      3.4                      pypi_0    pypi
importlib-metadata        6.8.0              pyha770c72_0    conda-forge
importlib_metadata        6.8.0                hd8ed1ab_0    conda-forge
intel-openmp              2023.1.0         h59b6b97_46319
ipykernel                 6.25.2             pyh60829e3_0    conda-forge
ipython                   8.16.1             pyh5737063_0    conda-forge
ipywidgets                8.1.1                    pypi_0    pypi
jedi                      0.19.1             pyhd8ed1ab_0    conda-forge
[... 119 lines cut ...]
tk                        8.6.12               h2bbff1b_0
toml                      0.10.2             pyhd3eb1b0_0
toolz                     0.12.0                   pypi_0    pypi
tornado                   6.3.3           py310h2bbff1b_0
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.11.2             pyhd8ed1ab_0    conda-forge
typing-extensions         4.8.0                hd8ed1ab_0    conda-forge
typing_extensions         4.8.0              pyha770c72_0    conda-forge
tzdata                    2023c                h04d1e81_0
urllib3                   2.0.6                    pypi_0    pypi
vc                        14.2                 h21ff451_1
[... cut ...]
xz                        5.4.2                h8cc25b3_0
yarl                      1.9.2                    pypi_0    pypi
zarr                      2.16.1                   pypi_0    pypi
zeromq                    4.3.4                h0e60522_1    conda-forge
zict                      3.0.0                    pypi_0    pypi
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.5                hd43e919_0
```

zm711, 2024-01-12:
@litzj01 could you try to update threadpoolctl? You can try:

```sh
conda update threadpoolctl
```
We want the version for that to be > 3.x.

litzj01, 2024-01-12:
I just ran `conda update threadpoolctl`, but it seems to not have worked properly. I now get an error trying to initialize spike interface (`import spikeinterface.full as si`) with the error` ImportError: Numba needs NumPy 1.25 or less`.

Here is the new `conda list`:
```
aiohttp                   3.8.6                    pypi_0    pypi
aiosignal                 1.3.1                    pypi_0    pypi
altair                    5.1.2                    pypi_0    pypi
asciitree                 0.3.3                    pypi_0    pypi
asttokens                 2.0.5              pyhd3eb1b0_0
async-timeout             4.0.3                    pypi_0    pypi
attrs                     23.1.0                   pypi_0    pypi
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
blas                      1.0                         mkl
bottleneck                1.3.5           py310h9128911_0
brotli                    1.0.9                h2bbff1b_7
brotli-bin                1.0.9                h2bbff1b_7
bzip2                     1.0.8                he774522_0
ca-certificates           2023.12.12           haa95532_0
cbor2                     5.5.1                    pypi_0    pypi
certifi                   2023.7.22                pypi_0    pypi
cffi                      1.16.0                   pypi_0    pypi
charset-normalizer        3.3.0                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
cloudpickle               3.0.0                    pypi_0    pypi
colorama                  0.4.6           py310haa95532_0
comm                      0.1.2           py310haa95532_0
contourpy                 1.0.5           py310h59b6b97_0
cryptography              41.0.5                   pypi_0    pypi
cuda-python               12.2.0                   pypi_0    pypi
cycler                    0.11.0             pyhd3eb1b0_0
cython                    3.0.0           py310h2bbff1b_0
dask                      2023.10.1                pypi_0    pypi
debugpy                   1.6.7           py310hd77b12b_0
decorator                 5.1.1              pyhd3eb1b0_0
distinctipy               1.2.3                    pypi_0    pypi
distributed               2023.10.1                pypi_0    pypi
exceptiongroup            1.0.4           py310haa95532_0
executing                 0.8.3              pyhd3eb1b0_0
fasteners                 0.19                     pypi_0    pypi
figurl                    0.2.18                   pypi_0    pypi
figurl-jupyter            0.2.8                    pypi_0    pypi
fonttools                 4.25.0             pyhd3eb1b0_0
freetype                  2.12.1               ha860e81_0
frozenlist                1.4.0                    pypi_0    pypi
fsspec                    2023.10.0                pypi_0    pypi
giflib                    5.2.1                h8cc25b3_3
glib                      2.69.1               h5dc1a3c_2
h5py                      3.9.0           py310hfc34f40_0
hdf5                      1.12.1               h51c971a_3
hdmf                      3.11.0                   pypi_0    pypi
herdingspikes             0.3.102                  pypi_0    pypi
icc_rt                    2022.1.0             h6049295_2
icu                       73.1                 h6c2663c_0
idna                      3.4                      pypi_0    pypi
importlib-metadata        6.8.0              pyha770c72_0    conda-forge
importlib_metadata        6.8.0                hd8ed1ab_0    conda-forge
intel-openmp              2023.1.0         h59b6b97_46320
ipykernel                 6.25.2             pyh60829e3_0    conda-forge
ipython                   8.20.0          py310haa95532_0
ipywidgets                8.1.1                    pypi_0    pypi
jedi                      0.18.1          py310haa95532_1
jinja2                    3.1.2                    pypi_0    pypi
joblib                    1.2.0           py310haa95532_0
jpeg                      9e                   h2bbff1b_1
jsonschema                4.19.2                   pypi_0    pypi
jsonschema-specifications 2023.7.1                 pypi_0    pypi
jupyter_client            8.6.0           py310haa95532_0
jupyter_core              5.5.0           py310haa95532_0
[... 114 lines cut ...]
toml                      0.10.2             pyhd3eb1b0_0
tomli                     2.0.1           py310haa95532_0
toolz                     0.12.0                   pypi_0    pypi
tornado                   6.3.3           py310h2bbff1b_0
tqdm                      4.66.1                   pypi_0    pypi
traitlets                 5.7.1           py310haa95532_0
typing-extensions         4.8.0                hd8ed1ab_0    conda-forge
typing_extensions         4.8.0              pyha770c72_0    conda-forge
tzdata                    2023d                h04d1e81_0
urllib3                   2.0.6                    pypi_0    pypi
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
wcwidth                   0.2.5              pyhd3eb1b0_0
wheel                     0.41.2          py310haa95532_0
widgetsnbextension        4.0.9                    pypi_0    pypi
xarray                    2023.9.0                 pypi_0    pypi
xz                        5.4.5                h8cc25b3_0
yarl                      1.9.2                    pypi_0    pypi
zarr                      2.16.1                   pypi_0    pypi
zeromq                    4.3.4                hd77b12b_0
zict                      3.0.0                    pypi_0    pypi
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h8cc25b3_0
zstd                      1.5.5                hd43e919_0
```

zm711, 2024-01-12:
That's weird because the numba and numpy are both still the same version in your environment... You could try to update threadpoolctl with pip instead 
```sh
pip install -U threadpoolctl
```

but if it keeps giving you the numba/numpy error I would consider downgrading numpy to 1.24. 

```sh
pip install numpy =1.24
```

litzj01, 2024-01-12:
Updating threadpoolctl worked, but it didn't fix the issue with numpy so I did have to downgrade it (I downgraded to numpy 1.25.0). That fixed the issue with initializing spike interface. 
I am trying to export the waveforms again which seems to be working. I'll keep you updated on how it goes. Thanks for your help!

zm711, 2024-01-16:
@litzj01, did this fix your issue? Just wanted to make sure we had solved the problem.

litzj01, 2024-01-22:
Yes, this seemed to have fixed the issue! Thanks for your help! Sorry for the late reply, I was without power for the week due to bad weather.

zm711, 2024-01-22:
Thanks for the response! Helps us keep track of things. And sorry about the power, hopefully the weather (and the waveform extracting) get a bit better!

PR #2413 (2024-01-22): Add version limit to `threadpoolctl`
See associated issue #2412, but base Anaconda ships with 2.2.0. 

The short story: < 3 can have some issues that can break parallel processing for users (see https://github.com/SpikeInterface/spikeinterface/issues/2401 and https://github.com/SpikeInterface/spikeinterface/issues/2310).

This PR just sets that limit at 3.0.0.

(Fixes #2412 , Fixes #2401, Fixes #2310)