SpikeInterface/spikeinterface #2406: New ISI violations based quality metric

jpv8888, 2024-01-12:
Hello, my name's Jack and I'm a PhD student in Mike Economo's lab at Boston University. Recently I derived an equation for predicting False Discovery Rate (FDR) from ISI violations in spike sorting that builds on previous work from Hill and Llobet. Chiefly, the prediction is now generalizable to any number of assumed contaminant neurons, and inhomogeneous spiking can be taken into account by looking at the firing rates over time of nearby clusters and inputting them into the equation. We also examine how effective ISI violations are in general as a source of information on cluster isolation, mainly finding that they are a much more robust indicator of overall sorting quality across a session or dataset than as a per cluster metric. The article is currently out for review, but the preprint is available here on bioRxiv:

https://www.biorxiv.org/content/10.1101/2023.12.21.572882v1.abstract

We would like to include our implementation in SpikeInterface; given its popularity and known effectiveness in the spike sorting community at the moment, we think including it here would be a great way to ensure people can easily take advantage of our new quality metric. It seems like it would fit logically into the preexisting quality metric structure as an alternative to Hill and Llobet's implementations:

https://spikeinterface.readthedocs.io/en/latest/modules/qualitymetrics/isi_violations.html

I reached out to Alessio Buccino about including the metric, and I was advised to open an issue here. Looking forward to helping in any way I can. 

Best,
Jack

samuelgarcia, 2024-01-12:
Hi Jack.
thank you for sharing this work, I will try to read this soon.
Very good idea to propose this inside spikeinterface.
I propose @DradeAW to review the the PR :)

alejoe91, 2024-01-29:
Hi @jpv8888 

Would you have time to tackle this and try to make a PR? 
We can also organize a chat and discuss about details :)

Let us know!

jpv8888, 2024-01-29:
Hi @alejoe91 

Yes, thank you for reaching out again, I will take a stab at it later this week. Organizing a chat would probably be good; I use slack with email jackv@bu.edu, but I can use any messaging platform. 

I haven't looked too closely at the SpikeInterface code, so sorry if this is a basic question, but for now I guess I would want to know is there an easy way to obtain for each cluster its ISI violation rate/fraction and a trial averaged PSTH or just firing rate over time? These are really the only two necessary pieces of information I would need to plug into the equation. 

Optionally, it would also be cool to restrict possible contaminating clusters input to the equation to those on the same or adjacent electrodes, although I'm not sure how difficult or easy that would be. Thanks!

zm711, 2024-06-19:
Sorry to ping you again. Is there still interest in implementing this?

jpv8888, 2024-06-26:
Hi Zach, no worries, thanks for reaching out again. Since making the original post, I've created a repository that can implement the method described in the preprint at: 

https://github.com/economolab/DCISIv

Susie (https://github.com/smp12304), an undergraduate with solid programming experience in our lab has just started working on an implementation of this code for SpikeInterface. We'll reach out once it's done and ready for testing and a pull request or if we have any questions. Thank you!

zm711, 2025-07-17:
@jpv8888, this fell off my radar sorry! Seems like it has been a year. Is there still interest here for your undergrad to add this?

zm711, 2025-08-21:
Sounds good Jack! No rush. We are always happy to support people so just let us know what we can do to help!