SpikeInterface/spikeinterface #2409: Issue Regarding the get_waveforms Method about version 0.99

sunjie454, 2024-01-17:
I use mountain_sort5 for data sorting, and then try to call the get_waveforms method to obtain waveforms for each unit. The following is a portion of my code along with usage instructions (indicating "The returned waveform (num_spikes, num_samples, num_channels)"). 
[image]
[image]
[image]
[image]
[image]
[image]

    In version 0.98, there isn't any problems, and I can retrieve waveforms from the channels corresponding to the unit. However, in version 0.99, a problem arises where the returned data shape is (num_spikes, num_samples, 3), instead of (num_spikes, num_samples, num_channels). Could someone please explain what the third dimension of the data returned by the get_waveforms method is in version 0.99, and how I can obtain waveforms for the channels corresponding to this unit?

samuelgarcia, 2024-01-17:
Here in the doc : https://spikeinterface.readthedocs.io/en/latest/modules/core.html?highlight=sparsity#sparsity

zm711, 2024-06-19:
Let us know if you still have trouble, I'll close this now.

closed 2024-06-19 by zm711