SpikeInterface/spikeinterface #2411: Cannot open Neuralynx data with Pegasus

Antho2422, 2024-01-17:
Hi all, 

I had the issue for a quite long time ago now but I get no answers from Neo side so I was wondering if I can maybe get some help from here  maybe :)

I think in the newest versions of Pegasus the import of Neuralynx data is impossible due to a wrong datetimeformat. Here is the link to my issue I posted on Neo's github : https://github.com/NeuralEnsemble/python-neo/issues/1352#issue-2008472760

Any help would be appreciated !

Cheers, 
Anthony

zm711, 2024-01-17:
I responded on the Neo side. I could maybe play around with it over there, but I don't have any data so I would definitely need to be provided some test files if I were going to work on it. And I'm not sure when I would have time to do it for sure.

Antho2422, 2024-01-17:
Thanks a lot I would be happy to help if needed on this part, I could give a try

zm711, 2024-03-08:
With Sam's recent patches is this working now @Antho2422? I assume you'll be at his workshop coming up so if you have more questions you can discuss there, but is this issue fixed now?