SpikeInterface/spikeinterface #2414: utf-8' codec can't decode byte 0x82 in position 80: invalid start byte

GabrielGaugain, 2024-01-18:
Dear all,

Trying to use SpikeInterface with python 3.8.18 and I have the following error when calling si.installed_sorters()
Do you know how to deal with this issue ?
Here is the Traceback:

`
UnicodeDecodeError                        Traceback (most recent call last)
Cell In[5], [line 1](vscode-notebook-cell:?execution_count=5&line=1)
----> [1](vscode-notebook-cell:?execution_count=5&line=1) si.installed_sorters()

File [d:\Users\ggaugain\Anaconda3\lib\site-packages\spikeinterface\sorters\sorterlist.py:59](file:///D:/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py:59), in installed_sorters()
     [56](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py?line=55) def installed_sorters():
     [57](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py?line=56)     """Lists installed sorters."""
---> [59](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py?line=58)     return sorted([s.sorter_name for s in sorter_full_list if s.is_installed()])

File [d:\Users\ggaugain\Anaconda3\lib\site-packages\spikeinterface\sorters\sorterlist.py:59](file:///D:/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py:59), in <listcomp>(.0)
     [56](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py?line=55) def installed_sorters():
     [57](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py?line=56)     """Lists installed sorters."""
---> [59](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/sorterlist.py?line=58)     return sorted([s.sorter_name for s in sorter_full_list if s.is_installed()])

File [d:\Users\ggaugain\Anaconda3\lib\site-packages\spikeinterface\sorters\external\hdsort.py:90](file:///D:/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/external/hdsort.py:90), in HDSortSorter.is_installed(cls)
     [88](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/external/hdsort.py?line=87) @classmethod
     [89](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/external/hdsort.py?line=88) def is_installed(cls):
---> [90](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/external/hdsort.py?line=89)     if cls.check_compiled():
     [91](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/external/hdsort.py?line=90)         return True
     [92](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/external/hdsort.py?line=91)     return check_if_installed(cls.hdsort_path)

File [d:\Users\ggaugain\Anaconda3\lib\site-packages\spikeinterface\sorters\basesorter.py:363](file:///D:/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py:363), in BaseSorter.check_compiled(cls)
    [355](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=354) shell_cmd = f"""
    [356](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=355) #!/bin/bash
    [357](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=356) if ! [ -x "$(command -v {cls.compiled_name})" ]; then
   (...)
    [360](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=359) fi
    [361](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=360) """
    [362](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=361) shell_script = ShellScript(shell_cmd)
--> [363](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=362) shell_script.start()
    [364](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=363) shell_script.wait()
    [365](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/basesorter.py?line=364) retcode = shell_script.wait()

File [d:\Users\ggaugain\Anaconda3\lib\site-packages\spikeinterface\sorters\utils\shellscript.py:91](file:///D:/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py:91), in ShellScript.start(self)
     [87](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=86) self._process = subprocess.Popen(
     [88](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=87)     cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT, bufsize=1, universal_newlines=True
     [89](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=88) )
     [90](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=89) with open(script_log_path, "w+") as script_log_file:
---> [91](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=90)     for line in self._process.stdout:
     [92](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=91)         script_log_file.write(line)
     [93](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=92)         if (
     [94](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=93)             self._verbose
     [95](file:///d%3A/Users/ggaugain/Anaconda3/lib/site-packages/spikeinterface/sorters/utils/shellscript.py?line=94)         ):  # Print onto console depending on the verbose property passed on from the sorter class

File [d:\Users\ggaugain\Anaconda3\lib\codecs.py:322](file:///D:/Users/ggaugain/Anaconda3/lib/codecs.py:322), in BufferedIncrementalDecoder.decode(self, input, final)
    [319](file:///d%3A/Users/ggaugain/Anaconda3/lib/codecs.py?line=318) def decode(self, input, final=False):
    [320](file:///d%3A/Users/ggaugain/Anaconda3/lib/codecs.py?line=319)     # decode input (taking the buffer into account)
    [321](file:///d%3A/Users/ggaugain/Anaconda3/lib/codecs.py?line=320)     data = self.buffer + input
--> [322](file:///d%3A/Users/ggaugain/Anaconda3/lib/codecs.py?line=321)     (result, consumed) = self._buffer_decode(data, self.errors, final)
    [323](file:///d%3A/Users/ggaugain/Anaconda3/lib/codecs.py?line=322)     # keep undecoded input until the next call
    [324](file:///d%3A/Users/ggaugain/Anaconda3/lib/codecs.py?line=323)     self.buffer = data[consumed:]

UnicodeDecodeError: 'utf-8' codec can't decode byte 0x82 in position 80: invalid start byte
`

Best

h-mayorquin, 2024-01-29:
Windows and old python it seems like this is a string eval issue. Try to re-write the script that's being executed by the pipe as it got corrupted somewhere.

zm711, 2024-02-06:
That is a spikeinterface script being string evaluated. So it is probably just better to update python versions right?

GabrielGaugain, 2024-02-14:
I did an update in my python distrib (now 3.11) and it still does not work (same behavior). Maybe due to windows I will take a look but it was to point out the issue

zm711, 2024-02-14:
I use spikeinterface with windows all the time so I don't think Windows per se is the problem. I'm wondering if this is related to an interaction between windows and hdsort. Do you have hdsort on your machine?

GabrielGaugain, 2024-02-14:
Nope 

> I use spikeinterface with windows all the time so I don't think Windows per se is the problem. I'm wondering if this is related to an interaction between windows and hdsort. Do you have hdsort on your machine?

Nope, is it needed ?

zm711, 2024-02-14:
It's not needed. But `is_installed` command iterates through the list of all possible sorters. If I'm not mistaken hdsort is the second in the list of all possible sorter so this command was failing early on in the list. If I run it on my computer where I don't have HDSort, it has been running fine. I was checking to see if it was failing because you had hdsort (with a bad install). But that doesn't  seem to be the problem. 

Windows 10 or 11?

GabrielGaugain, 2024-02-14:
Windows 10 so far

zm711, 2024-02-14:
For your Windows path which shell are you using?

GabrielGaugain, 2024-02-15:
Why do you mean ? The python path that I set in my windows path variables ?

zm711, 2024-02-15:
@alejoe91, should we document that the code base requires use of `cmd prompt` instead of other shells. Windows now has three shells built in (cmd, powershell, and terminal). Not sure if you want to try to handle all possible Windows shells or just `cmd prompt`.

But if this fixed your issues @GabrielGaugain, I would just suggest always using cmd prompt for now. :)

alejoe91, 2024-02-15:
@zm711 yes that is a good idea! Anything that would minimize these kind of "tricky" and platform-specific problems is largely appreciated!

zm711, 2024-02-15:
Sounds good. I'll do a doc PR soon :)

closed 2024-03-08 by samuelgarcia