SpikeInterface/spikeinterface #2415: spikeinterface.toolkit

m-bellv, 2024-01-18:
Hello,

Whenever I try: import spikeinterface.toolkit as st I get the error ModuleNotFoundError: No module named 'spikeinterface.toolkit'.
I installed spikeinterface using pip install spikeinterface[full], but have since tried pip install spikeinterface[full, widgets] as well as git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .
and the error persists. I am using Ubuntu 20.04.6 LTS, but have also tried installing on the Digital Research Alliance of Canada narval server with the same issue.

Thanks for any help!

alejoe91, 2024-01-18:
Hi,

The `spikeinterface.toolkit` has been first deprecated and then removed. Where did you find documentation about 
it?

What code snippet are you running?

m-bellv, 2024-01-18:
Oh I apologize, that was an oversight on my part. I was trying to use it to exclude artifact periods of the recording, so the recording_rm_artifact = st.remove_artifacts(recording, stimulation_trigger_frames, ms_before=, ms_after=), from https://spikeinterface.readthedocs.io/en/0.90.0/module_toolkit.html. Is there an alternative function I can use?

alejoe91, 2024-01-18:
you should use the latest documentation:  https://spikeinterface.readthedocs.io/en/0.99.1

The toolkit has basically been split into other submodules, so you can find the same function in the `spikeinterface.preprocessing` module: https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.remove_artifacts

closed 2024-01-18 by m-bellv