SpikeInterface/spikeinterface #2417: Speed up LFP signal extraction

YinJiapeng, 2024-01-19:
I am extracting LFP signals from raw recording data (SpikeInterface version 0.98.0). My code is:
    
```
    recording_lfp = spre.bandpass_filter(recording, freq_min=1, freq_max=300)  # 1st step
    recording_lfp = spre.resample(recording_lfp, 1000) # 2nd step
  
    traces0 = recording_lfp.get_traces(segment_index=0)  # 3rd step
```
    
    I found the 1-2 step is quite fast. But the 2rd step is slow. 
    
    How can I use parallel processing for the 3rd step? Or is there any other method to speed up this process?

samuelgarcia, 2024-01-19:
Hi Yin.
Step 1 and 2 are lazy this is why they are fast.
For step 3 you can use `save()` function with folder or memory and then have the parralelisation.

```python

recording_lfp = spre.bandpass_filter(recording, freq_min=1, freq_max=300)  # 1st step
recording_lfp = spre.resample(recording_lfp, 1000) # 2nd step

 # 3rd step

# in memory
recording_saved = recording_lfp.save(format="memory", n_jobs=12, chunk_duration="1s", progress_bar=True)
# or in folder if not enough memory
recording_saved = recording_lfp.save(format="binary", folder="/tempory/folder", n_jobs=12, chunk_duration="1s", progress_bar=True)

traces0 = recording_saved.get_traces(segment_index=0)
```

zm711, 2024-03-25:
@YinJiapeng, let us know if you have any more questions. Feel free to open a new issue if needed. :)

closed 2024-03-25 by zm711