SpikeInterface/spikeinterface #2419: waveclus not working

ZoeChen96, 2024-01-19:
Dear,

I was trying a published single-channel dataset [(dataset3)](https://le.ac.uk/csn/data) with WaveClus. However, I can't do it either in spikeinterface or Waveclus GUI. 

1) In spikeinterface, what I did is:
```
import spikeinterface.sorters as ss
import spikeinterface.extractors as se
sortingGT_all = se.read_waveclus(file_path = '/UCLA_data/UCLA_data/times_CSC4.mat')
recording = se.read_neuralynx(folder_path = '/UCLA_data/UCLA_data')
ss.WaveClusSorter.set_waveclus_path('/gitrepository/wave_clus')
sorting_sorter = ss.run_sorter('waveclus', recording, verbose=True)
```

And the error is:
```
/imec/users/chen14/.local/lib/python3.8/site-packages/tqdm/auto.py:22: TqdmWarning: IProgress not found. Please update jupyter and ipywidgets. See https://ipywidgets.readthedocs.io/en/stable/user_install.html
  from .autonotebook import tqdm as notebook_tqdm
Setting WAVECLUS_PATH environment variable for subprocess calls to: /imec/other/macaw/chen14/gitrepository/wave_clus
Warning! The recording is already filtered, but waveclus filter is enabled
Num. channels = 1, Num. timepoints = 61000192, duration = 31.720378979335017 minutes
Running waveclus in /imec/other/macaw/chen14/pysimulator/development notebooks/waveclus_output/sorter_output..
RUNNING SHELL SCRIPT: /imec/other/macaw/chen14/pysimulator/development notebooks/waveclus_output/sorter_output/run_waveclus.sh
                            < M A T L A B (R) >
                  Copyright 1984-2020 The MathWorks, Inc.
              R2020a Update 8 (9.8.0.1873465) 64-bit (glnxa64)
                              February 3, 2022
                            < M A T L A B (R) >
                  Copyright 1984-2020 The MathWorks, Inc.
              R2020a Update 8 (9.8.0.1873465) 64-bit (glnxa64)
                              February 3, 2022
To get started, type doc.
To get started, type doc.
For product information, visit www.mathworks.com.
For product information, visit www.mathworks.com.
1 of 1 'spikes' files done.
1 of 1 'spikes' files done.
Caught "std::exception" Exception message is:
Message Catalog MATLAB:license was not loaded from the file. Please check file location, format or contents
Caught "std::exception" Exception message is:
Message Catalog MATLAB:license was not loaded from the file. Please check file location, format or contents
```

ZoeChen96, 2024-01-19:
2. in waveclus GUI, I followed the instructions and load the `CSC4.Ncs` data with the button
[image]
and the error is:
```
Error using wave_clus>load_data_button_Callback
(line 230)
Less than 15 spikes detected
Error in gui_mainfcn (line 95)
        feval(varargin{:});
Error in wave_clus (line 63)
    gui_mainfcn(gui_State, varargin{:});
Error while evaluating UIControl Callback.
```

ZoeChen96, 2024-01-19:
PS: I have an extra question about the [dataset](https://le.ac.uk/csn/data)]. It has a wave_clus formatted sorting file, which I can read with `sortingGT_all = se.read_waveclus(file_path = '/UCLA_data/UCLA_data/times_CSC4.mat')`. Can it be seen as a ground truth, or is it just a wave_clus result?

Maybe this question is out-of-scope of the SI project, but I see waveclus project is already archived so I can't put issues there, sorry for putting my question here...

zm711, 2024-01-22:
@ZoeChen96, waveclus is archived as you pointed out @ferchaure may be able to help, but I'm not sure how much time they dedicate to the project any more. (@alejoe91 should we move waveclus to the legacy portion of the docs? to highlight that its not being fully supported any more?)

ferchaure, 2024-01-22:
Hi @ZoeChen96, About the error I would check the sign of the detection. 

I'm guessing that is a manually curated result... it can be used as ground truth in some applications. But probably the best way to find out that type of thing is writing to someone currently in the lab

@zm711 I left academia and I'm not sure of the current development around waveclus, maybe Hernan Rey will take care / update it. Waveclus has a lot of users, but not the particularly tech-savvy kind.... probably a difficult crowd for the current spikeinterface

zm711, 2024-03-25:
I'm going to close this. @ZoeChen96 please take Fernando's advice and reach out to someone in the lab for further waveclu trouble shooting . But if you run into any other issues feel free to open a new issue :)

closed 2024-03-25 by zm711