SpikeInterface/spikeinterface #2422: Spike timepoints for individual units?

taningh86, 2024-01-21:
Is there a built-in function to extract spike timepoints for each unit? I am looking to get spike timepoints to find units that fire at the same timepoints or have correlated firing of some sort.

alejoe91, 2024-01-21:
Yes of course! `sorting.get_unit_spike_train(unit_id)`

You can read more about the `Sorting` object here: https://spikeinterface.readthedocs.io/en/latest/modules/core.html#sorting

taningh86, 2024-01-22:
Thanks Alessio! This is very helpful!

taningh86, 2024-01-22:
Is it possible to get it for all units? It looks like 'get_unit_spike_train' specifically gets for one unit at a time and putting a range don't seem to work.

alejoe91, 2024-01-22:
`sorting.get_all_spike_trains()` (https://spikeinterface.readthedocs.io/en/latest/api.html?highlight=Get_all_spike_trains#spikeinterface.core.BaseSorting.get_all_spike_trains)

samuelgarcia, 2024-01-22:
Hi,
get_all_spike_trains() is deprecated use insteadto_spike_vector()

alejoe91, 2024-01-22:
@samuelgarcia 

It's not deprecated and it's the only function to return a dictionary of spike times...

samuelgarcia, 2024-01-22:
No we decided to depreacte it.

https://spikeinterface.readthedocs.io/en/latest/api.html?highlight=Get_all_spike_trains#spikeinterface.core.BaseSorting.get_all_spike_trains

https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/core/basesorting.py#L457

alejoe91, 2024-01-22:
It can be convenient though no?

taningh86, 2024-01-22:
Thanks guys! Looks like to_spike_vector() is the one togo for version 0.99.1.0

zm711, 2024-02-06:
I'll close this for now, but we could open a new issue to debate the deprecation if necessary. @taningh86 feel free to reach out if new things come up.

closed 2024-02-06 by zm711