SpikeInterface/spikeinterface #2424: Is there a function to extract correlation between units in their spike timing?

taningh86, 2024-01-22:
Hi.  In other words, is there a function to list units that have similar temporal firing pattern. I am able to get the spike timepoints of each unit using the 'get_unit_spike_train' function. I can try to find out spike timepoints for all the other units and write up a code to find similar timepoints of spikes betwen different units and to what degree they are similar. However, that can take a bit longer on my end to do so. I wonder if there's a function that already exists in SI to find correlation of spike firing between units and cluster them based on correlation in their temporal firing. If that function does not exist then it will be very helpful to have one.
Thanks!

alejoe91, 2024-01-22:
Hi,

For this kind of downstream analysis, I recommend checking out the [Elephant](https://elephant.readthedocs.io/en/latest/modules.html#spike-trains) package, which has a lot of tools and functions for spike train statistics and correlation!

closed 2024-01-22 by alejoe91

taningh86, 2024-01-22:
Thanks a lot! This has pretty much everything I was looking for!