SpikeInterface/spikeinterface #2425: Different errors in importing TDT and Blackrock files.

00Ziling, 2024-01-22:
Hi!
I'm starting to work with SpikeInterface and want to use it to process data recorded from TDT system and Blackrock system. I'm having issues reading both the TDT and Blackrock files. I've checked the issues part here to try to find the solution of it. However, I couldn't fix it so far. Could you please help me with it? Thanks!

There is my code and error information  for the TDT data:
[image]
[image]

And the code and error information for the Blackrock data:
[image]
[image]

h-mayorquin, 2024-01-29:
Do you have data that you can share with us so we can test? 

For TDT the folder that you are using does not seem to contain the corresponding file.
Blackrock takes a file path not a folder path.

jesusdpa1, 2024-02-13:
Hi! I am also experiencing problems trying to load our tdt files. I didn't observe this behavior when using ```from neo.rawio import TdtRawIO```

```py
# Data struct from tdt reader
SPO2	[struct]
DEMA	[struct]
DEMB	[struct]
BrtR	[struct]
BPrS	[struct]
ErOX	[struct]
HRae	[struct]
Temp	[struct]
BPrr	[struct]
DA_w	[struct]
DB_w	[struct]
```

```py
#%%
from pathlib import Path
import spikeinterface.full as si
print(f"SpikeInterface version: {si.__version__}")
#%%
import matplotlib.pyplot as plt
import numpy as np
from pathlib import Path
import os

import warnings
warnings.simplefilter("ignore")

# %matplotlib widget
%matplotlib inline

#%%
home = Path.home()
dir_name = r'Dropbox (UFL)\Data\02-09-24_8898-1_testSubject-HD EMG electrodes 5th try'
dir_path = home.joinpath(dir_name)
# %%
dir_list = list(dir_path.glob('*/'))
print(dir_list)
# %%
full_raw_rec = si.read_tdt(dir_list[0], stream_name='DEMA')
# %%
import tdt
# %%
data_ = tdt.read_block(dir_list[1])

```

```py
IndexError                                Traceback (most recent call last)
File \ephys-robot\tests\spike_test.py:2
      1 # %%
----> 2 full_raw_rec = si.read_tdt(dir_list[0], stream_name='DEMA')

File miniconda3\envs\roboz\Lib\site-packages\spikeinterface\extractors\neoextractors\tdt.py:34, in TdtRecordingExtractor.__init__(self, folder_path, stream_id, stream_name, block_index, all_annotations)
     32 def __init__(self, folder_path, stream_id=None, stream_name=None, block_index=None, all_annotations=False):
     33     neo_kwargs = self.map_to_neo_kwargs(folder_path)
---> 34     NeoBaseRecordingExtractor.__init__(
     35         self,
     36         stream_id=stream_id,
     37         stream_name=stream_name,
     38         block_index=block_index,
     39         all_annotations=all_annotations,
     40         **neo_kwargs,
     41     )
     42     self._kwargs.update(dict(folder_path=str(Path(folder_path).absolute())))

File miniconda\envs\roboz\Lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py:187, in NeoBaseRecordingExtractor.__init__(self, stream_id, stream_name, block_index, all_annotations, use_names_as_ids, **neo_kwargs)
    158 def __init__(
    159     self,
    160     stream_id: Optional[str] = None,
   (...)
    165     **neo_kwargs: Dict[str, Any],
...
--> 179 self._global_t_start = self._seg_t_starts[0]
    181 # signal channels EVTYPE_STREAM
    182 signal_streams = []

IndexError: list index out of range

```

jesusdpa1, 2024-02-13:
After investigating a little more I found that the loader is asking for the ```.tev``` file 

```sh
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---            2/9/2024 12:05 PM            138 Notes.txt
-a---            2/9/2024 11:54 AM           2379 StoresListing.txt
-a---            2/9/2024 12:05 PM         294149 testSubject-240209_00_baseline.Tbk
-a---            2/9/2024 12:05 PM         340224 testSubject-240209_00_baseline.Tdx
-a---            2/9/2024 12:05 PM     3951132280 testSubject-240209_00_baseline.tev
-a---            2/9/2024 12:05 PM          59015 testSubject-240209_00_baseline.tin
-a---            2/9/2024 11:54 AM             23 testSubject-240209_00_baseline.tnt
-a---            2/9/2024 12:05 PM       20630720 testSubject-240209_00_baseline.tsq

```

```py
data_path = list(dir_list[1].glob('*.tev'))
data_ = se.read_tdt(data_path[0], stream_name="b'DA_w'")
```

```py
# output
TdtRecordingExtractor: 16 channels - 24.4kHz - 1 segments - 14,714,880 samples 
                       602.72s (10.05 minutes) - float32 dtype - 898.12 MiB
```

h-mayorquin, 2025-09-09:
Closing this, feel free to re-open if there is a pending issue.