SpikeInterface/spikeinterface #2428: Could we reuse plotting function made for WaveformExtractor for Templates

yger, 2024-01-23:
While the addition of the Templates object is great, the problem is that it can not make use of all the plotting functions that have already been built for templates, but assuming a waveformexctrator as input. Should we modify these plotting functions accordingly, or discuss about a possible mecanism to close the gap between the two objects?

yger, 2024-01-23:
Actully, I think that to be consistent, the get_all_templates() function of the waveform_extractor should return a Template object, don't you think?

samuelgarcia, 2024-01-23:
This will be done with the SortingAnalyzer/SortingResult implementation.

yger, 2024-01-24:
Ok, but what is the plan? Because to move forward with hybrid project, I would need to use functions made for WaveformExtractor (like in template_tools.py for example). Do you plan of extending all signatures with something like waveform_extractor_or_templates as you are doing currently in #2410, or are you planning to make Templates and WaveformExtractor inherit from a common object? Conceptually, they share quite a lot of attributes, so I don't know

zm711, 2024-03-25:
@yger, did you still want plotting for Templates or is the SortingAnalyzer fine and we can close this now?

yger, 2024-03-25:
I would like to be able to have the plot_unit_template widget working when only fast_templates is computed....

alejoe91, 2024-07-02:
This has been implemented in https://github.com/SpikeInterface/spikeinterface/pull/2856

closed 2024-07-02 by alejoe91