SpikeInterface/spikeinterface #2429: recording.set_probe gives an error

Ashkees, 2024-01-23:
Hello,

I'm new to spike interface, so my apologies if I'm making a silly error, or if this isn't the correct channel for asking questions.

I've loaded my 96-channel intan data using
```
recording = si.load_binary(filepath)
```
and I would like to set the probe.  I've created a ProbeGroup object that consists of 32 channels from a linear probe and 64 channels from a 4-shank probe. I've saved it as a .json and  I've tried 
```
recording.set_probe(probegroup) 
```
but this throws the following error:
```
AssertionError: must give Probe
```
What could I be doing incorrectly?  I've attached the probegroup .json.

Thanks!
Ashley
[SI_4shank_linear.json](https://github.com/SpikeInterface/spikeinterface/files/14025767/SI_4shank_linear.json)

alejoe91, 2024-01-23:
Hi @Ashkees !

To set a probe group you should use the `set_probegroup(probegroup)` instead :)

Ashkees, 2024-01-23:
Ah, yes, that was silly.  That worked!  thank you =)

Ashkees, 2024-01-23:
I didn't think to look for another method because the first argument to set_probe is probe_or_probegroup so I assumed it could take a probegroup

alejoe91, 2024-01-23:
No worries! It's much less common to use probe groups instead of probes, so that's probably why it was a bit harder to find in tutorials/docs ;)

closed 2024-01-23 by alejoe91