SpikeInterface/spikeinterface #2433: How to split the waveform object into several waveforms object for a single recording and sorting ?

babougr, 2024-01-24:
Hello, 

I have a question about a process I'm currently doing with my data : 

My goal is to have template metrics for my waveforms for 3 periods on a single recording. 

I'm doing the following steps : 

- Loading the recording

- Splitting recording into 3 subrecordings using the following function (according to the timings of my markers): 

```python
recording1 = recording.frame_slice(0, int(marker_dict['Vent_Off'] * recording.get_sampling_frequency()))
recording2 = recording.frame_slice(int(marker_dict['Vent_Off'] * recording.get_sampling_frequency()), int(marker_dict['Vent_On'] * recording.get_sampling_frequency()))
recording3 = recording.frame_slice(int(marker_dict['Vent_On'] * recording.get_sampling_frequency()), recording.get_total_samples())
``` 

- Splitting the associated sorting as follow : 

```python
sorting1  = si.core.split_sorting(parent_sorting=sorting,recording_or_recording_list=[recording1])
sorting2= si.core.split_sorting(parent_sorting=sorting,recording_or_recording_list=[recording2])
sorting3 =si.core.split_sorting(parent_sorting=sorting,recording_or_recording_list=[recording3])
``` 

- Then I register each corresponding recordings to the associated sortings : 

```python
sorting1.register_recording(recording=recording1)
sorting2.register_recording(recording=recording2)
sorting3.register_recording(recording=recording3)
``` 

- And finally I extract every waveforms using the sub recordings and sub sorting as follow : 
```python
waveforms1 = si.extract_waveforms(recording=sorting1._recording, sorting=sorting1, max_spikes_per_unit=None, mode='memory', ms_before=2, ms_after=2, dtype='float64')
waveforms2 = si.extract_waveforms(recording=sorting2._recording, sorting=sorting2, max_spikes_per_unit=None, mode='memory', ms_before=2, ms_after=2, dtype='float64')
waveforms3 = si.extract_waveforms(recording=sorting3._recording, sorting=sorting3, max_spikes_per_unit=None, mode='memory', ms_before=2, ms_after=2, dtype='float64')
``` 

My goal is to have 3 waveforms extractor object so I can visualize and compute metrics on the 3 periods of interests from a single segment recording. 

The issue is that I have currently weird shaped waveforms templates when getting the templates of the 2nd and 3rd waveforms extractor object. 

I would like to know where the issue could be : 

- Is it coming from my data ? 
- Is my code wrong and if so, is there an easier and more reliable way to do this ? 

Thank you, 

Cheers, 
Aymard

alejoe91, 2024-01-24:
Hi @babougr 

Almost there! Instead of the `si.split_sorting` function (BTW, you don't need `si.core`), you should use:
```
sorting1 = sorting.frame_slice(0, int(marker_dict['Vent_Off'] * recording.get_sampling_frequency()))
...
```
(same as recording)

Also, you don't need to register the recording. You can simply run `extract_waveforms(recording1, sorting1)`.

The reason is that the `split_function` only looks for start and end frame. After slicing the recording, its start frame will always be internally reset to 0.

Let us know if it works!

babougr, 2024-01-24:
Hi @alejoe91 

I didn't know that we could also split the sorting with frame_sclice, it's great !

 It's working now, Thanks a lot !! 

Cheers, 
Aymard.

zm711, 2024-01-29:
Looks like this is complete. @babougr feel free to open more issues if anything else comes up!

closed 2024-01-29 by zm711