SpikeInterface/spikeinterface #2434: getting spike time for all good units

smri07, 2024-01-24:
hello, 

I have applied some threshold and calculated the quality metrics. As an output I get all 'my good units'. 
The next thing, I want to do is to extract the spike time for all my good units.
my question is: do we have any function which can extract the spike time for all the good units at once.

I tried to define a (python) function where i looped over the unitsID columns in the quality metrics dataframe and use get_unit_spike_train but it doesn't work because get_unit_spike_train takes int as an input. 

sorry for my naïve question, I am a beginner 
Thank you

zm711, 2024-01-24:
Hey @smri07,

The documentation has a great tutorial on how to just get the "good units" (just adapt for your preferred qc metrics):

https://spikeinterface.readthedocs.io/en/latest/modules_gallery/qualitymetrics/plot_4_curation.html#sphx-glr-modules-gallery-qualitymetrics-plot-4-curation-py

Then you can do:

```python
spike_vector = curated_sorting.to_spike_vector()
```

 spike_vector is a structured numpy array ("sample_index", "unit_index", "segment_index") with all spikes

If you're doing something completely differently than the tutorial then it would be helpful if you could post your script here so we can see how you're deciding on your "good unit" ids so we can help.

smri07, 2024-01-24:
[image]

@zm711 Thank you for the response, I did the same thing but getting this error

smri07, 2024-01-24:
metrics = si.compute_quality_metrics(waveform)
keep_mask = (metrics['amplitude_cutoff']< 0.1) & (metrics['isi_violations_ratio']<1) & (metrics['presence_ratio'] > 0.9)
keep_unit_ids = keep_mask[keep_mask].index.values
keep_unit_ids = [unit_id for unit_id in keep_unit_ids]

zm711, 2024-01-24:
Could you show:

```python
print(sorting.get_unit_ids())
print(keep_unit_ids)
```

smri07, 2024-01-24:
[image]
I do get all my unit ids and curated unit ids

zm711, 2024-01-24:
Let's check the assert message that is occurring inside the curation by doing the following:

```python
assert all(unit_id in sorting.get_unit_ids() for unit_id in keep_unit_ids)
```
Then if that fails try:

```python
bad_ids = []
for unit_id in keep_unit_ids:
    if unit_id not in sorting.get_unit_ids():
        bad_ids.append(unit_id)
print(bad_ids)
```

smri07, 2024-01-24:
[image]
The first command didn't work..!!

zm711, 2024-01-24:
How are the metrics data and the sorting related? Because we basically see that your qc metrics have a bunch of units which don't exist in your sorting. Are you sure the metrics you're trying to use come from this particular sorting and not another sorting?

smri07, 2024-01-24:
They come from the same sorting

zm711, 2024-01-24:
Next test then would be to check your `waveform_extractor`. Whatever you ran the metrics on (in the tutorial it's called `we`)

so do

```python
assert all(unit_id in sorting.get_unit_ids() for unit_id in waveform_extractor.unit_ids())
```

alejoe91, 2024-01-24:
Thanks for helping @zm711 

Might be the usual int/str problem!

@smri07 what is the output of `print(sorting.unit_ids.dtype)` and `print(type(keep_unit_ids[0]))`?

zm711, 2024-01-24:
Thanks @alejoe91 I was just getting ready to tap you in.

smri07, 2024-01-24:
[image]

alejoe91, 2024-01-24:
Mmm they are both integers! Can you share as Zach mentioned the script until quality metrics computation? Is it possible that the sorting object is not the correct one?

samuelgarcia, 2024-01-29:
Hi @smri07
why not something like

```python
some_unit_ids = 
curated_sorting = sorting.select_unit_ids(some_unit_ids)

```