SpikeInterface/spikeinterface #2437: Absolute path to recording isn't saved when using export_to_phy

ikharitonov, 2024-01-24:
In case of using _export_to_py_ function and not copying the binary recording (copy_binary=False), "None" gets saved in _params.py_ instead of the absolute path. Seems to happen because _waveform_extractor.recording_ does not necessarily have to be a _BinaryRecordingExtractor_ (e.g. in my case it is a _CommonReferenceRecording_) when checking for `isinstance(waveform_extractor.recording, BinaryRecordingExtractor)`, hence the _rec_path_ does not get extracted.

https://github.com/SpikeInterface/spikeinterface/blob/c2f18bd536437e9f500a4e24f7a0f3773bd79d3a/src/spikeinterface/exporters/to_phy.py#L144C55-L144C79

alejoe91, 2024-01-24:
That is the correct behavior. Sonce you recording is not binary, if you want the `.dat` file to be available to Phy you need `copy_binary=True`, and this will save the processed data in the Phy folder

ikharitonov, 2024-01-24:
But I am using a binary recording in my case, just with some preprocessing applied on top (e.g. common reference)

zm711, 2024-01-24:
@ikharitonov,

The preprocessing steps are lazy. So your preprocessed recording object does not yet have have the preprocessing applied to it, but instead keeps an internal list of all the preprocessing which should be applied. The `copy_binary=True` will take all the preprocessing and apply it in parallel as it is writing the binary file (this can take a while because it is no longer lazy) which will give you a new binary file with the preprocessing applied. Your old binary file is still the same without any of the preprocessing applied to it yet. So in this case you want to `copy_binary=True` to make sure the binary file you are getting has all of the operations that you've performed on it included. 

Does that distinction make sense?

ikharitonov, 2024-01-25:
@zm711 that does make sense, thank you!

closed 2024-01-25 by ikharitonov