SpikeInterface/spikeinterface #2440: Error running SpykingCircus2 clustering

Antho2422, 2024-01-26:
Hi @yger,

I have a question about the clustering process in spyking circus. I noticed that sometimes when I run the sorter on small periods of data or channels with a few peaks I get the following error. I assumed that it was coming from a lack of spikes to perform a clustering but I would like to make sure it is not coming from something else. 

[image]

Thank you, 
Anthony

yger, 2024-01-26:
If this is a small recording, can you share it with me (alongside with the analysis script)?

Antho2422, 2024-01-26:
Of course @yger , here is a link to download the dataset : https://instituteicm-my.sharepoint.com/:f:/g/personal/anthony_pinto_icm-institute_org/EpiPint_OABJiQNFPG2Q-44B7cmBfe_iV1CJOI_RrHLZKg?e=XCZgWL

It is already filtered using a bandpass filter between 300 and 6000Hz with dtype64 so I did not put apply preprocessing in the run_sorter function. 

The sorting is integrated in a pipeline so it is difficult to give you the full code but I made the following code that is replicating the process to reproduce the error. Here is the code :

```python
from spikeinterface.sorters import run_sorter
import spikeinterface as si

recording = si.load_extractor(r'C:\Users\anthony.pinto\Documents\dataset_fail_clustering\recording.zarr')
params = si.sorters.get_default_sorter_params('spykingcircus2')
params['waveforms']['max_spikes_per_unit'] = 500
params['detection']['peak_sign'] = 'pos'
params['detection']['detect_threshold'] = 3.5779258099871694
params['apply_preprocessing'] = False
sorting = run_sorter('spykingcircus2', recording, output_folder='test_run_sorting', **params)
``` 

The weird value of the detection threshold is determined automatically so don't mind it. I didn't tweak the other parameters appart from those in the above code, maybe I should but I'm not sure of what each parameter is doing so that is why I didn't change a lot of the params. 

Thank you, 
Anthony

yger, 2024-01-26:
So on my machine, with the main branch, the code is working fine. This is both a good and a bad news :-( Can you tell me the version of hdbscan that you have?

Antho2422, 2024-01-26:
Argh, yes ofc here is my hdbscan version : 

[image]

Note also that I get this working with both Windows and Linux.

yger, 2024-01-29:
When you say "you get this working", you mean that the code is crashing? I have the sam  hdbscan version, so maybe send me the full version for the envs (pip list)

Antho2422, 2024-01-29:
Yes sorry for the formulation, I mean the code crashes. 

Here is my pip list : 

```python
Package                   Version      Editable project location
------------------------- ------------ ------------------------------------------
aiohttp                   3.9.1
aiohttp-retry             2.8.3
aiosignal                 1.3.1
altair                    5.0.1
amqp                      5.2.0
annotated-types           0.6.0
antlr4-python3-runtime    4.9.3
antropy                   0.1.5
anyio                     4.1.0
appdirs                   1.4.4
argon2-cffi               23.1.0
argon2-cffi-bindings      21.2.0
arrow                     1.3.0
asciitree                 0.3.3
asttokens                 2.4.1
async-timeout             4.0.3
asyncssh                  2.14.1
atomicwrites              1.4.1
atpublic                  4.0
attrs                     23.1.0
audioread                 3.0.1
autoreject                0.4.3
backcall                  0.2.0
beautifulsoup4            4.12.2
billiard                  4.2.0
black                     23.11.0
bleach                    6.1.0
cachetools                5.3.2
cbor2                     5.4.6
celery                    5.3.6
certifi                   2023.11.17
cffi                      1.16.0
cfgv                      3.4.0
chardet                   5.2.0
charset-normalizer        3.3.2
click                     8.1.7
click-didyoumean          0.3.0
click-plugins             1.1.1
click-repl                0.3.0
cloudpickle               3.0.0
colorama                  0.4.6
colorcet                  3.0.1
coloredlogs               15.0.1
comm                      0.2.0
configobj                 5.0.8
contourpy                 1.2.0
coverage                  7.3.2
cryptography              41.0.7
cycler                    0.12.1
Cython                    0.29.36
dacite                    1.8.1
dask                      2023.11.0
debugpy                   1.8.0
decorator                 5.1.1
defusedxml                0.7.1
dictdiffer                0.9.0
diskcache                 5.6.3
distlib                   0.3.7
distributed               2023.5.1
distro                    1.8.0
dpath                     2.1.6
dulwich                   0.21.6
dvc                       3.33.4
[... 245 lines cut ...]
tzdata                    2023.3
uri-template              1.3.0
urllib3                   2.1.0
vine                      5.1.0
virtualenv                20.24.7
visions                   0.7.5
viziphant                 0.3.0
voluptuous                0.14.1
watchdog                  3.0.0
wcwidth                   0.2.12
webcolors                 1.13
webencodings              0.5.1
websocket-client          1.6.4
wheel                     0.40.0
widgetsnbextension        4.0.9
wordcloud                 1.9.2
xmltodict                 0.13.0
yarl                      1.9.3
yasa                      0.6.3
ydata-profiling           4.5.1
zarr                      2.16.1
zc.lockfile               3.0.post1
zict                      3.0.0
zipp                      3.17.0
```

wliWanru, 2024-02-06:
Hi Anthony,

I encountered a similar problem with my single-channel data today.

I had installed SpikeInterface several months ago, and in the version I had 0.99.1 directly installed via pip, the following lines of code in `sortingcomponents/clustering/random_projections.py`

```
projections = np.random.randn(num_chans, d["nb_projections"])
projections -= projections.mean(0)
projections /= projections.std(0)
```
resulted in all NaN values in the 'projections' array, which then caused a ValueError.
Then I found that this issue has been fixed in the current version of the SpikeInterface repository, as seen in this commit: https://github.com/SpikeInterface/spikeinterface/commit/93fbc55a29d769020b4835156dd1bfe094e8296c

I noticed that you also dealing with single-channel data, and I'm curious if you experienced a similar problem due to the same reason. If so, the main branch version works fine!

Antho2422, 2024-02-06:
Hi @wliWanru, 

Thanks for this update, I will try to update spikeinterface and see if this works. Thank you

alejoe91, 2024-02-06:
Note that version 0.100.0nwas just released today!

zm711, 2025-03-18:
Since this is over one year old and there have been updates to SI and SC2 I'll close this issue and feel free to open a new issue if you have other problems with SI or SC2.

closed 2025-03-18 by zm711