SpikeInterface/spikeinterface #2444: Adding exclude_filename argument to NeuralynxRecordingExtractor

Antho2422, 2024-01-29:
Hi all, 

I would like to suggest a possible feature that could be helpful and one possible implementation. I may be wrong because I'm not very familiar with Neo but I'm giving a try.

The point of this feature is to take advantage of the argument "exclude_filename" of this Neo function:
```python
class neo.rawio.NeuralynxRawIO(dirname='', filename='', exclude_filename=None, keep_original_times=False, **kargs)
``` 
This argument can be helpful to avoid some files contained in a Neuralynx folder. In my case I have one Events.nev file that I want to skip because it's causing me some troubles but I cannot modify the folder directly. 

When looking more into the SpikeInterface code I found that we can maybe add the parameter to the NeuralynxRecordingExtractor __init__ ? One possible implementation would be the following (Note that I did not test that, it's just a first possible solution I guess) :

```python
class NeuralynxRecordingExtractor(NeoBaseRecordingExtractor):
    """
    Class for reading neuralynx folder

    Based on :py:class:`neo.rawio.NeuralynxRawIO`

    Parameters
    ----------
    folder_path: str
        The file path to load the recordings from.
    stream_id: str, default: None
        If there are several streams, specify the stream id you want to load.
    stream_name: str, default: None
        If there are several streams, specify the stream name you want to load.
    all_annotations: bool, default: False
        Load exhaustively all annotations from neo.
    exlude_filename: list[str], default: None
        List of filename to exclude from the loading.
    """

    mode = "folder"
    NeoRawIOClass = "NeuralynxRawIO"
    name = "neuralynx"

    def __init__(self, folder_path, stream_id=None, stream_name=None, all_annotations=False, exlude_filename=None):
        neo_kwargs = self.map_to_neo_kwargs(folder_path, exlude_filename)
        NeoBaseRecordingExtractor.__init__(
            self, stream_id=stream_id, stream_name=stream_name, all_annotations=all_annotations, **neo_kwargs
        )
        self._kwargs.update(dict(folder_path=str(Path(folder_path).absolute())))

    @classmethod
    def map_to_neo_kwargs(cls, folder_path, exlude_filename):
        neo_kwargs = {"dirname": str(folder_path), 'exclude_filename': exlude_filename}
        return neo_kwargs
``` 

Thank you,

Cheers, 
Anthony

alejoe91, 2024-01-29:
Hi Anthony!

If the neo argument is already existing we should definitely add it as you suggested! Can you make a PR? :)

Antho2422, 2024-01-29:
Hi @alejoe91, 

Done just here : #2445

zm711, 2024-01-31:
Completed by #2445.

closed 2024-01-31 by zm711