SpikeInterface/spikeinterface #2448: separate folders for saved clustering objects when using 2 probes

Ashkees, 2024-01-30:
Hello,

I'm working with a dataset that has two probes (one linear probe in one region, one multi-shank in another region).  I'm defining them in ProbeInterface as two probes and then grouping them together before applying them to the recording.  Then I run the sorter using run_sorter_by_property.  It saves the results in two subfolders, which I can then load separately.

My code looks like this:
```
raw_rec = full_raw_rec.set_probegroup(probes)
sorting_SC2 = si.run_sorter_by_property('spykingcircus2', raw_rec, grouping_property='group',
                                working_folder=base_folder / 'results_SC2',
                                verbose=True, job_kwargs=job_kwargs)

# to load from saved
sorting_SC2_loaded = si.read_sorter_folder(base_folder / "results_SC2" / '0')
sorting_SC2_loaded_1 = si.read_sorter_folder(base_folder / "results_SC2" / '1')

```

What I'd like to know is whether there is a way to 'glue' back together the two folders when I load them?  Because all the subsequent steps (waveform extraction, exportation to phy) are done on two separate objects and I'd like it to be over only one object per recording, if possible.

Thanks so much!
Ashley

alejoe91, 2024-01-30:
Hi Ashley!

You can use the `aggregate_units` function as shown [here](https://spikeinterface.readthedocs.io/en/latest/modules/core.html?highlight=aggregate_units#manipulating-objects-slicing-aggregating)

In your case, you woule do:
```
sorting_SC2 = si.aggregate_units([sorting_SC2_loaded, sorting_SC2_loaded_1])
```

A better solution, in my opinion, is to save the `sorting_SC2` directly:

```
sorting_SC2 = sorting_SC2.save(folder="aggregated_sorting")

sorting_SC2_loaded = si.load_extractor("aggregated_sorting")
```
This option is better because it also stores the properties of the object. In this case, the `run_sorter_by_property` will add a property `group` to each unit with the recording it was found on.

Ashkees, 2024-01-30:
Fantastic!  Thank you so much for getting back to me quickly with my basic questions!

alejoe91, 2024-01-30:
No worries!

closed 2024-01-30 by alejoe91